raspberrypi/pico-sdk #1937: Poor RISC-V `pico_riscv_gcc_zcb_zcmp` performance

matsobdev, 2024-09-20:
Now I have a performance issue. Compilers like CORE-V are huge in size, so decided to trim them down a bit. One thing is to leave only default library, by deleting all extra ones inside:
```
~/corev-openhw-gcc-ubuntu2004-20240530/riscv32-corev-elf/lib
~/corev-openhw-gcc-ubuntu2004-20240530/lib/gcc/riscv32-corev-elf/14.1.0
```
did a trick. BTW same for ARM.

Today I've got my Pico 2, but noticed something before. Having only default lib, binaries had different size compared to vanilla compiler. Now tested both, and when using [pico_riscv_gcc](https://github.com/raspberrypi/pico-sdk/blob/master/cmake/preload/toolchains/pico_riscv_gcc.cmake) everything's fine.

But for vanilla CORE-V and [pico_riscv_gcc_zcb_zcmp](https://github.com/raspberrypi/pico-sdk/blob/master/cmake/preload/toolchains/pico_riscv_gcc_zcb_zcmp.cmake) it compiles fine, but performance is poor. I have some air mouse from an accelerometer at 6664 Hz, and RISC-V is a bit slower on Pico 2 than M0+ on Pico, but it was still able to keep up with a stream of data and moving average and stuff. Creating monster `pico_riscv_gcc_zcb`, that is `pico_riscv_gcc_zcb_zcmp` without `_zcmp` behaved similar.

Removing extra libraries from CORE-V (same with xPack) resolved an issue. Now both `pico_riscv_gcc_zcb_zcmp` and `pico_riscv_gcc_zcb` are performing better than `pico_riscv_gcc`. It is latest 14.1.0 GCC CORE-V. Default library is `rv32imac/ilp32`. Avoiding shenanigans with tuning compiler, changing:
```
set(PICO_COMMON_LANG_FLAGS " -march=rv32ima_zicsr_zifencei_zba_zbb_zbs_zbkb_zca_zcb_zcmp -mabi=ilp32")
```
to
```
set(PICO_COMMON_LANG_FLAGS " -march=rv32imac_zicsr_zifencei_zba_zbb_zbs_zbkb_zca_zcb_zcmp -mabi=ilp32")
```
resolves poor performance issue for [pico_riscv_gcc_zcb_zcmp](https://github.com/raspberrypi/pico-sdk/blob/master/cmake/preload/toolchains/pico_riscv_gcc_zcb_zcmp.cmake).  Same works for xPack.

PS. I propose to create `pico_riscv_gcc_zcb`. For CORE-V 14.1.0 GCC it performed better than `pico_riscv_gcc_zcb_zcmp`. First had 2,7% performance boost, latter 0,19% in comparison with `pico_riscv_gcc`

ilg-ul, 2024-09-21:
> Removing extra libraries from CORE-V (same with xPack) resolved an issue.

This is intriguing. 

Could you do a verbose link with the xPack toolchain (the official release, with all libraries) and check what library was actually used, then with your trimmed xPack, and compare the results?

matsobdev, 2024-09-21:
I'll take care of it for sure later in sunday, but now what I have from the past: on Windows, CORE-V required leaving `rv32ia` - `pico_riscv_gcc_zcb_zcmp` and `rv32iac` - `pico_riscv_gcc`. [RISC-V](https://www.embecosm.com/resources/tool-chain-downloads/#riscv-stable) required leaving `rv32iac` - `pico_riscv_gcc` and since it doesn't have `zca` that `pico_riscv_gcc_zcb` wasn't working there. It was determined by try and error of `ilp32's`. Without them compilation was terminated with error.

Just for the record - what is previous is about Ubuntu 20.04. Last word for now - it seems like it can pick up from multiple matching libraries, whichever is available.

matsobdev, 2024-09-22:
Back to vanilla Pico SDK. I don't really know, how to check, which library is picked up, but removed all libraries including a default one in that two directories and xPack has four `ilp32's`:
```
rv32i
rv32ia
rv32iac
rv32im
```
so replacing one by one as a default one, so only default library was present. Compilation was successfully every single time, program was working every time on Pico 2 but inly when `rv32im` was present, `pico_riscv_gcc` and `pico_riscv_gcc_zcb` (vanilla Pico SDK apart from this one, since xPack has no `zcmp` so far) performance was at a predicted level similar to Pico (it is adding up values for moving average in, atan2f() and division in proportion of 4000/2/2). Obviously, original default `rv32imac` works and performs good as stated before. I guess it has all the letters to pick up from :D But question is, should it be a specific flavour of a library or can it selectively use `ima` from `imac`

ilg-ul, 2024-09-22:
> I don't really know, how to check, which library is picked up

Try adding `-Wl,-v,--verbose` to the linker command.

matsobdev, 2024-09-22:
Tried `add_compile_options(-Wl,-v,--verbose)` inside `CMakeLists.txt`, nothing happened, but according to `*.elf.map` files:

- `pico_riscv_gcc` picks up default `rv32imac`, [pico_riscv_gcc.txt](https://github.com/user-attachments/files/17090973/pico_riscv_gcc.txt)
- `pico_riscv_gcc_zcb` picks up `rv32ia`, [pico_riscv_gcc_zcb.txt](https://github.com/user-attachments/files/17090970/pico_riscv_gcc_zcb.txt)

Files renamed due to Github requirements. Ori Pico SDK and xPack.

ilg-ul, 2024-09-22:
> Tried add_compile_options(-Wl,-v,--verbose) inside CMakeLists.txt nothing happened

Sure it does nothing, since you have to add it to `target_link_options()`.

Otherwise, if you use `rv32ima`, the resulting code will not use the compressed instructions, the projects will be larger and probably slower.

matsobdev, 2024-09-22:
Thanks for CMake tip, was struggling with that one. More I dig into it less I know. But leaving inside xPack `rv32ima_zicsr_zifencei_zba_zbb_zbs_zbkb_zca_zcb` only as a default and binary size is 46,9 kB for `pico_riscv_gcc_zcb`, 47,4 kB for `pico_riscv_gcc`. Leaving only `rv32imac_zicsr_zifencei_zba_zbb_zbs_zbkb` and 47,0 kB and 47,5 kB. Why there are two of them if both are working and performing good :D Too much for today...

matsobdev, 2024-09-23:
Quickest run so far, was CORE-V GCC 14.1.0 with `rv32imac_zicsr_zifencei_zba_zbb_zbs_zbkb` libraries and `pico_riscv_gcc_zcb_zcmp`.

Wren6991, 2024-11-12:
I think in the longer term we would like to move away from CORE-V; really that CMake option exists because we used CORE-V to build the C parts of the RISC-V bootrom, and there was no upstream GCC support at the time.

ilg-ul, 2024-11-12:
>  move away from CORE-V

As maintainer of a RISC-V toolchain from the very early days ([xPack RISC-V GCC](https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/releases/), I had to fight a lot of issues with merging various forks.

A 0.8% speed increase or a 0.5K/47K size reduction are not worth the effort.

Therefore I suggest you generally stick close to the official GNU GCC distribution.

Wren6991, 2024-11-12:
> 0.5K/47K size reduction are not worth the effort.

For the ROM this was the right choice, but for all other purposes yes, this is vestigial and I would like to remove it now that upstream Zcmp support is in a better place.