raspberrypi/pico-sdk #1938: CYW43 pins are hard coded

sfe-SparkFro, 2024-09-20:
These should really be set in the board definition file, so other board variants can use other pins.

https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c#L24-L30

kilograham, 2024-09-20:
see #1816

closed 2024-09-20 by kilograham

sfe-SparkFro, 2024-09-21:
Doh! Should have looked at the develop branch, thank you!

Ok, so if a variant uses different GPIO pins, is it as simple as `#define`ing the following in the board header?

```
CYW43_DEFAULT_PIN_WL_REG_ON
CYW43_DEFAULT_PIN_WL_DATA_OUT
CYW43_DEFAULT_PIN_WL_DATA_IN
CYW43_DEFAULT_PIN_WL_HOST_WAKE
CYW43_DEFAULT_PIN_WL_CLOCK
CYW43_DEFAULT_PIN_WL_CS
```

sfe-SparkFro, 2024-09-21:
Also, are there any restrictions on which pins can be used, or is any combination of pins valid?

kilograham, 2024-09-21:
> Doh! Should have looked at the develop branch, thank you!
>
> Ok, so if a variant uses different GPIO pins, is it as simple as `#define`ing the following in the board header?
>
> ``` [...]

yes

kilograham, 2024-09-21:
> Also, are there any restrictions on which pins can be used, or is any combination of pins valid?

question for @peterharperuk

peterharperuk, 2024-10-11:
Sorry, I saw this question, then lost the issue. In theory there's no restriction on what gpios you can use. But we've only tested the code with data in, out and wake set to the same pin (the configuration pico_w uses), so using different pins for these might not work. On, clock and cs must be set to their own, different pins. Assuming this is about RM2 I've had some issues running the pio at clock div of 2 when connected to a device with wires, so you may have to slow it down to 3 with  CYW43_PIO_CLOCK_DIV_INT=3. Email me if you want more details.

sfe-SparkFro, 2024-10-11:
Understood, thank you!