raspberrypi/pico-sdk #1946: GCC LTO not working

matsobdev, 2024-09-23:
Adding:
```
set(CMAKE_INTERPROCEDURAL_OPTIMIZATION TRUE)
```
into `CMakeLists.txt` to activate LTO results in:
```
PICO_SDK_PATH is /home/mateush/pico-sdk-2.0.0
Target board (PICO_BOARD) is 'pico2'.
Using board configuration from /home/mateush/pico-sdk-2.0.0/src/boards/include/boards/pico2.h
Pico Platform (PICO_PLATFORM) is 'rp2350-arm-s'.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m33_gcc'
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/mateush/Dysk RAM/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mateush/Dysk RAM/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mateush/Dysk RAM/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
-- Found Python3: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter 
TinyUSB available at /home/mateush/pico-sdk-2.0.0/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/mateush/pico-sdk-2.0.0/lib/btstack
cyw43-driver available at /home/mateush/pico-sdk-2.0.0/lib/cyw43-driver
lwIP available at /home/mateush/pico-sdk-2.0.0/lib/lwip
mbedtls available at /home/mateush/pico-sdk-2.0.0/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mateush/Dysk RAM/test_rp2350_nowszy/build
[81/81] Linking CXX executable hog_kbd.elf
FAILED: hog_kbd.elf 
: && "/home/mateush/Dysk RAM/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-g++" -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -g -O3 -DNDEBUG -flto -fno-fat-lto-objects -Wl,--build-id=none -Wl,-Map=hog_kbd.elf.map --specs=nosys.specs -Wl,--wrap=__ctzdi2 -Wl,--wrap=__aeabi_dadd -Wl,--wrap=__aeabi_ddiv -Wl,--wrap=__aeabi_dmul -Wl,--wrap=__aeabi_drsub -Wl,--wrap=__aeabi_dsub -Wl,--wrap=__aeabi_cdcmpeq -Wl,--wrap=__aeabi_cdrcmple -Wl,--wrap=__aeabi_cdcmple -Wl,--wrap=__aeabi_dcmpeq -Wl,--wrap=__aeabi_dcmplt -Wl,--wrap=__aeabi_dcmple -Wl,--wrap=__aeabi_dcmpge -Wl,--wrap=__aeabi_dcmpgt -Wl,--wrap=__aeabi_dcmpun -Wl,--wrap=__aeabi_i2d -Wl,--wrap=__aeabi_l2d -Wl,--wrap=__aeabi_ui2d -Wl,--wrap=__aeabi_ul2d -Wl,--wrap=__aeabi_d2iz -Wl,--wrap=__aeabi_d2lz -Wl,--wrap=__aeabi_d2uiz -Wl,--wrap=__aeabi_d2ulz -Wl,--wrap=__aeabi_d2f -Wl,--wrap=sqrt -Wl,--wrap=cos -Wl,--wrap=sin -Wl,--wrap=tan -Wl,--wrap=atan2 -Wl,--wrap=exp -Wl,--wrap=log -Wl,--wrap=ldexp -Wl,--wrap=copysign -Wl,--wrap=trunc -Wl,--wrap=floor -Wl,--wrap=ceil -Wl,--wrap=round -Wl,--wrap=sincos -Wl,--wrap=asin -Wl,--wrap=acos -Wl,--wrap=atan -Wl,--wrap=sinh -Wl,--wrap=cosh -Wl,--wrap=tanh -Wl,--wrap=asinh -Wl,--wrap=acosh -Wl,--wrap=atanh -Wl,--wrap=exp2 -Wl,--wrap=log2 -Wl,--wrap=exp10 -Wl,--wrap=log10 -Wl,--wrap=pow -Wl,--wrap=powint -Wl,--wrap=hypot -Wl,--wrap=cbrt -Wl,--wrap=fmod -Wl,--wrap=drem -Wl,--wrap=remainder -Wl,--wrap=remquo -Wl,--wrap=expm1 -Wl,--wrap=log1p -Wl,--wrap=fma -Wl,--wrap=cosf -Wl,--wrap=sinf -Wl,--wrap=tanf -Wl,--wrap=atan2f -Wl,--wrap=expf -Wl,--wrap=logf -Wl,--wrap=ldexpf -Wl,--wrap=copysignf -Wl,--wrap=truncf -Wl,--wrap=floorf -Wl,--wrap=ceilf -Wl,--wrap=roundf -Wl,--wrap=sincosf -Wl,--wrap=asinf -Wl,--wrap=acosf -Wl,--wrap=atanf -Wl,--wrap=sinhf -Wl,--wrap=coshf -Wl,--wrap=tanhf -Wl,--wrap=asinhf -Wl,--wrap=acoshf -Wl,--wrap=atanhf -Wl,--wrap=exp2f -Wl,--wrap=log2f -Wl,--wrap=exp10f -Wl,--wrap=log10f -Wl,--wrap=powf -Wl,--wrap=powintf -Wl,--wrap=hypotf -Wl,--wrap=cbrtf -Wl,--wrap=fmodf -Wl,--wrap=dremf -Wl,--wrap=remainderf -Wl,--wrap=remquof -Wl,--wrap=expm1f -Wl,--wrap=log1pf -Wl,--wrap=fmaf -Wl,--wrap=malloc -Wl,--wrap=calloc -Wl,--wrap=realloc -Wl,--wrap=free "-Wl,-L/home/mateush/Dysk RAM/test_rp2350_nowszy/build" -Wl,--script=/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_crt0/rp2350/memmap_no_flash.ld -nostartfiles -Wl,-z,max-page-size=4096 -Wl,--gc-sections -Wl,--no-warn-rwx-segments -Wl,--wrap=sprintf -Wl,--wrap=snprintf -Wl,--wrap=vsnprintf -Wl,--wrap=printf -Wl,--wrap=vprintf -Wl,--wrap=puts -Wl,--wrap=putchar -Wl,--wrap=getchar CMakeFiles/hog_kbd.dir/main.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_stdlib/stdlib.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_gpio/gpio.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2350/pico_platform/platform.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_platform_panic/panic.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/hardware_claim/claim.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_sync/sync.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_sync_spin_lock/sync_spin_lock.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_irq/irq.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_irq/irq_handler_chain.S.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_sync/sem.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_sync/lock_core.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_sync/mutex.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_sync/critical_section.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_time/time.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_time/timeout_helper.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_timer/timer.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_util/datetime.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_util/pheap.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/common/pico_util/queue.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_uart/uart.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_clocks/clocks.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_pll/pll.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_vreg/vreg.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_watchdog/watchdog.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_ticks/ticks.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_xosc/xosc.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_divider/divider.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_runtime/runtime.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_runtime_init/runtime_init.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_runtime_init/runtime_init_clocks.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_runtime_init/runtime_init_stack_guard.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_bootrom/bootrom.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_bootrom/bootrom_lock.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/hardware_boot_lock/boot_lock.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_divider/divider_compiler.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_double/double_math.c.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_double/double_aeabi_dcp.S.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_double/double_fma_dcp.S.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_double/double_sci_m33.S.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0.0/src/rp2_common/pico_double/double_conv_m33.S.obj CMakeFiles/hog_kbd.dir/home/mateush/pico-sdk-2.0. [... cut ...]
/home/mateush/Dysk RAM/arm-gnu-toolchain-13.3.rel1-x86_64-arm-none-eabi/bin/arm-none-eabi-objcopy: '/home/mateush/Dysk RAM/test_rp2350_nowszy/build/hog_kbd.elf': No such file
ninja: build stopped: subcommand failed.
```

It does work on LLVM tho, and performance increase is quite substantial. Test program with some float and atan2f() and lots of non floating point math. Release build type and array of final results is printed out after bencharking is done, just for parts of program not to be optimised out. LTO LLVM binary takes about 70-something% time of non-LTO LLVM to complete.

lurch, 2024-09-23:
See also the discussions in #97

matsobdev, 2024-09-24:
Thanks, I should search more. So closing if already exist.

closed 2024-09-24 by matsobdev