raspberrypi/pico-sdk #1947: call to calloc() deadlocks with clang/picolibc

geurtv, 2024-09-24:
With clang 18.1.3 (from [LLVM-embedded-toolchain-for-Arm](https://github.com/ARM-software/LLVM-embedded-toolchain-for-Arm)) the `calloc` function in picolibc (so that's `__real_calloc`) calls `malloc`, which then deadlocks because `__wrap_malloc` tries to lock the mutex already owned by `__wrap_calloc`:

```
calloc == __wrap_calloc:
    lock mutex
    call __real_calloc

__real_calloc:
    call malloc

malloc == __wrap_malloc:
    lock mutex
    << deadlock >>
```

My work-around is to replace `__wrap_calloc`'s implementation with a call to `__wrap_malloc` and then a `memset`:

```c
void *WRAPPER_FUNC(calloc)(size_t count, size_t size) {
    size *= count;
    void* ptr = WRAPPER_FUNC(malloc)(size);
    memset(ptr, 0, size);
    return ptr;
}
```

NOTE 1: `malloc`+`memset` to replace `calloc` cannot be used because clang (and also gcc) will replace that code with a call to `calloc`.
NOTE 2: picolibc sources say `malloc` already 'sets to zero', but it's probably not smart to rely on that.

geurtv, 2024-09-25:
This issue also applies to `realloc` for which I don't have a work-around, except not locking the mutex in `__wrap_realloc`.

geurtv, 2024-10-02:
Safest fix might be to change `malloc_mutex` into a recursive mutex. Then it will always just work, also if an unlikely future version of newlib changes its implementation. Otherwise something like this can be done:

```c
#ifndef __PICOLIBC__

auto_init_mutex(malloc_mutex);
#define malloc_mutex_enter_blocking mutex_enter_blocking
#define malloc_mutex_exit           mutex_exit

#else

auto_init_recursive_mutex(malloc_mutex);
#define malloc_mutex_enter_blocking recursive_mutex_enter_blocking
#define malloc_mutex_exit           recursive_mutex_exit

#endif
```

kilograham, 2024-11-20:
yeah; i had thought about recursive mutex, but it means you go very wrong if you nest in an IRQ

kilograham, 2024-11-21:
merged into develop

closed 2024-11-21 by kilograham