raspberrypi/pico-sdk #1955: Add a Github action in CI to check for a working bazel build

cramertj, 2024-09-25:
Adding a minimal check that does something like:

* Do the bazel setup https://pigweed.dev/docs/get_started/github_actions.html
* Grab HEAD of Picotool
* run ./tools/run_all_bazel_checks.py --picotool-dir=./tmp/picotool

Would help to track when the Bazel build is broken and needs fixing, as well as helping after-the-fact to figure out what ranges have a known-working Bazel build.

armandomontanez, 2024-10-04:
I put together a presubmit in #1973 that covers the following:
* Pico SDK Bazel build covering RP2040, RP2350, and a few extra configuration flavors (clang, Pico W) for macOS, Linux, and Windows.
* Host build of the few compatible tests macOS, Linux, and Windows.
* Pico SDK + Picotool build at `HEAD` of develop branch for macOS, Linux, and Windows.
* Pico SDK + Picotool build at current minimum BCR version of Picotool (i.e. oldest backwards-compatible release) for macOS, Linux, and Windows.
* Various other checks (check to ensure all files are enumerated in the Bazel build, ensure configuration options available in CMake are available in Bazel).

Currently, the macOS and Linux builds are passing. Windows is hitting some issues when building Picotool (unsurprisingly). The "other" check is failing right now since new Pico W-related configuration options were added that I haven't hooked up yet.

A few checks I'd also like to add eventually:
* Verify submodule pins match pins in `MODULE.bazel`.
* Verify project version strings match those in CMake.

kilograham, 2024-11-18:
moving to 2.2.0 as this is now partially done in 2.1.0

lurch, 2024-11-19:
@armandomontanez I think these

> A few checks I'd also like to add eventually:
> 
> * Verify submodule pins match pins in `MODULE.bazel`.
> * Verify project version strings match those in CMake.

were both done in #2039 , right?

closed 2024-11-19 by lurch