raspberrypi/pico-sdk #1963: Raspberry Pi Pico HID Joystick Limited to 6 Axes

Talha-Babar, 2024-09-30:
Summary
I'm using the Raspberry Pi Pico as a USB HID joystick with 8 axes using the TinyUSB stack, but it has only 6.

typedef struct TU_ATTR_PACKED
{
  int8_t  x;         ///< Delta x  movement of left analog-stick
  int8_t  y;         ///< Delta y  movement of left analog-stick
  int8_t  z;         ///< Delta z  movement of right analog-joystick
  int8_t  rz;        ///< Delta Rz movement of right analog-joystick
  int8_t  rx;        ///< Delta Rx movement of analog left trigger
  int8_t  ry;        ///< Delta Ry movement of analog right trigger
  uint8_t hat;       ///< Buttons mask for currently pressed buttons in the DPad/hat
  uint32_t buttons;  ///< Buttons mask for currently pressed buttons
}hid_gamepad_report_t;

Is there a limitation in TinyUSB or the Pico regarding support for more than 6 axes?
How can I achieve full support for 8 axes in the HID joystick?

Environment
Microcontroller: Raspberry Pi Pico

Any help would be greatly appreciated.

lurch, 2024-09-30:
ping @hathach

kilograham, 2024-09-30:
This is definitely a TinyUSB (https://github.com/hathach/tinyusb) question ... I imagine you need a different type of report

You could try asking on forums here too https://forums.raspberrypi.com/viewforum.php?f=143

closed 2024-09-30 by kilograham