raspberrypi/pico-sdk #1975: Error in the hardward/pwm.h

DarkElvenAngel, 2024-10-05:
There's an error in the SDK 2.0.0 version 

https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/hardware_pwm/include/hardware/pwm.h#L658

There needs to be a `, ""` added
```
/tmp/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h: In function 'pwm_set_irq_mask_enabled':
/tmp/pico-sdk/src/rp2_common/hardware_pwm/include/hardware/pwm.h:658:55: error: expected ',' before ')' token
     static_assert(PWM_IRQ_WRAP_1 == PWM_IRQ_WRAP_0 + 1);
                                                       ^
                                                       ,
```
I solved this by changing the line to 
``` 
   static_assert(PWM_IRQ_WRAP_1 == PWM_IRQ_WRAP_0 + 1, "");
```
This only happen when using the RP2350

lurch, 2024-10-06:
This seems to be a duplicate of #1788 ? (which was fixed on the `develop` branch by https://github.com/raspberrypi/pico-sdk/commit/424e21ec0793219e5b24f1566f3cbcf4263f5919 )

kilograham, 2024-11-11:
duplicate

closed 2024-11-11 by kilograham