raspberrypi/pico-sdk #1977: Core1 not starting correctly

rbryson74, 2024-10-07:
I was having some trouble getting some multicore code running. So I added some serial uart feedback to see what was happening. It appeared as if Core1 wouldn't start it's loop code without a soft reset immediately after bootup. I was able to replicate this with a clean project and minimal code. This is the 2.0 SDK in VSCode and the SDK plugin. This is also just a Pico1. I have not tried it on any of my PicoW. 

If the 5ms sleep and core1 reset are commented out or removed, core1 will never get to looping. It will say "Core1 up" but never says "Looping now" or "Hello from core1".

With the sleep and reset kept in, core1 seems to start just fine. 

multicore_test.c
`#include <stdio.h>
#include <stdbool.h>
#include "pico/stdlib.h"
#include "pico/multicore.h"

void core1_entry () {
    printf("Core1 Up\n\r");
    bool looping = false;
    while(1) {
        if (!looping) {
            printf("looping now\n\r");
            looping = true;
        }
        printf("Hello from core1!\n");
        sleep_ms(1007);
    }
}

int main()
{
    stdio_init_all();
    printf("\n\n\n\n**********boot start*************\n\n\n\n");
    printf("Core0 started\n");
    sleep_ms(5);
    multicore_reset_core1();
    sleep_ms(10);
    multicore_launch_core1(core1_entry);

    while (true) {
        printf("Hello, world!\n");
        sleep_ms(10262);
    }
}`

CMakeLists.txt (was generated by the plugin)
`# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NEVER EDIT THE NEXT LINES for Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.0.0)
set(toolchainVersion 13_2_Rel1)
set(picotoolVersion 2.0.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(multicore_test C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_executable(multicore_test multicore_test.c )

pico_set_program_name(multicore_test "multicore_test")
pico_set_program_version(multicore_test "0.1")

# Modify the below lines to enable/disable output over UART/USB
pico_enable_stdio_uart(multicore_test 1)
pico_enable_stdio_usb(multicore_test 0)

# Add the standard library to the build
target_link_libraries(multicore_test
        pico_stdlib
        pico_multicore)

# Add the standard include files to the build
target_include_directories(multicore_test PRIVATE
  ${CMAKE_CURRENT_LIST_DIR}
  ${CMAKE_CURRENT_LIST_DIR}/.. # for our common lwipopts or any other standard includes, if required
)

pico_add_extra_outputs(multicore_test)`

rbryson74, 2024-10-08:
And.... now I can't get it to replicate the issue. Let me poke at this a bit more. I should know soon if this in an actual issue or not.

rbryson74, 2024-10-09:
I've updated the original post to the minimum amount of code that will replicate this on my system. However, it only shows the issue after a SWD flash finishes. If I do a hard restart by resetting the power, it boots up just fine.

lurch, 2024-10-09:
How are you doing the SWD flash? Perhaps the SWD flash isn't resetting Core 1 properly, which is why it only works correctly after a "hard restart"?
Does https://github.com/raspberrypi/pico-feedback/issues/153 help?

rbryson74, 2024-10-15:
The SWD flash is being done via the "Flash" task setup by the Pico SDK Plugin. It is using OpenOCD. The command line is as follows:

```openocd.exe -s C:\Users\rbryson74/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c adapter speed 5000; program "c:/Sync/code/tft_pio_dma/build/tft_pio_dma.elf" verify reset exit```

However, once I got to the pico-feedback link you provided there was a comment for another issue ([https://github.com/raspberrypi/pico-feedback/issues/248](url)) that basically makes my submission an exact duplicate. 

However, the "flash" task only uses OpenOCD and there is no "monitor" command since that seems to be for GDB instead. 

So, I'm not sure how the command needs to be modified.

peterharperuk, 2024-10-15:
I always load images via gdb. I'll have to test loading via openocd.

kilograham, 2024-11-18:
I suspect this is a debug scenario where core 1 is not reset by the debugger on launch; we should consider just always resetting core 1 before launching it.

That said, i'm also curious if there is something peculiar to VS Code, so assigning to @will-v-pi for now

will-v-pi, 2024-11-19:
I can't reproduce this myself - using the current version of the VS Code extension (0.17.1) with the code you've provided it runs fine for me with the `sleep_ms(5)` commented or uncommented. Could you try updating to the latest version of the extension and creating a new project, to see if the problem still occurs?

will-v-pi, 2024-11-19:
This looks like an OpenOCD problem not a VS Code problem - I can reproduce this issue, where loading and resetting directly from OpenOCD (ie Flash task in VS Code) only prints "Core1 Up", but when loading using gdb over OpenOCD (ie debugging in VS Code) it runs fine. My recommendation would be to use debugging in VS Code to load the program for now, rather than using the Flash task.

> `openocd.exe -s C:\Users\rbryson74/.pico-sdk/openocd/0.12.0+dev/scripts -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c adapter speed 5000; program "c:/Sync/code/tft_pio_dma/build/tft_pio_dma.elf" verify reset exit`

This is the openocd command that seems to be causing the issues, but I'm not sure why, so assigning back to @kilograham as I don't think it's a VS Code issue

liamfraser, 2024-11-19:
Moving to 2.2.0 as sounds like an openocd issue