raspberrypi/pico-sdk #1980: No such file or directory: #include "hardware/dma.h"

rspber, 2024-10-09:
Hi everybody,

Is there already any solution to this ?

[proc] Executing command: /usr/bin/cmake --build /usr/public/tsdesktop/pico-sdk/build --config Release --target all -j 18 --
...
[build] [  3%] Building CXX object examples/menu1/CMakeFiles/menu1.dir/usr/public/tsdesktop/rp2040/spi/RP2040_TFT_SPI.cpp.obj
[build] /usr/public/tsdesktop/rp2040/spi/RP2040_TFT_SPI.cpp:15:10: fatal error: hardware/dma.h: No such file or directory
[build]    15 | #include "hardware/dma.h"
[build]       |          ^~~~~~~~~~~~~~~~
[build] compilation terminated.

It occures on fedora 39
 with eabi 13 and 14.
 cmake 3.27.7.
 downgrade cmake unavailable.

lurch, 2024-10-09:
Sounds like a missing `target_link_libraries` in your `CMakeLists.txt` ?

rspber, 2024-10-09:
Thanks, that was it.

closed 2024-10-09 by rspber