raspberrypi/pico-sdk #1990: Unable to build pico-sdk : Incompatible picotool installation found

all2coolnick, 2024-10-19:
I have just installed pico-sdk and picotool, and built picotools using Terminal on Mac OS 12.7.5 using the commands given in the RP getting-started-with-pico.pdf
No errors during installation.
picotools seems to work fine. I added the location of the instal to $PATH and if I type 'picotool version' I get the response...
`picotool v2.0.0 (Darwin, AppleClang-14.0.0.14000029, Release)`

However when I try to build the pico-sdk using cmake .. from the newly created build folder, I get the following out with a fatal error.

```
Defaulting platform (PICO_PLATFORM) to 'rp2040' since not specified.
Defaulting target board (PICO_BOARD) to 'pico' since not specified.
Using board configuration from /Users/nicholasgillard/git/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
-- Defaulting build type to 'Release' since not specified.
Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
CMake Error at tools/CMakeLists.txt:125 (message):
  Incompatible picotool installation found: Requires version 2.0.0, you have
  version unknown

  Update your installation, or set PICOTOOL_FORCE_FETCH_FROM_GIT to download
  and build the correct version
Call Stack (most recent call first):
  src/cmake/on_device.cmake:33 (pico_init_picotool)
  src/rp2040/boot_stage2/CMakeLists.txt:57 (pico_add_dis_output)
  src/rp2040/boot_stage2/CMakeLists.txt:101 (pico_define_boot_stage2)
```

I've deleted everything and done this again but still get the same error. Why can the build not identify that the installed picotool is the latest version 2.0.0?

matsobdev, 2024-10-20:
For me, standalone `picotool` compilation without:
```-D CMAKE_INSTALL_PREFIX=LOCATION_DIR_OF_PICOTOOL_BINARY -D PICOTOOL_FLAT_INSTALL=1```
gave some error. It was running on its own, but SDK was unhappy about it.

all2coolnick, 2024-10-20:
Thanks for the reply and sorry to be dense but can you clarify in what context and sequence to use the suggested commands? I’m unfamiliar with -D.

will-v-pi, 2024-10-21:
> I have just installed pico-sdk and picotool, and built picotools using Terminal on Mac OS 12.7.5 using the commands given in the RP getting-started-with-pico.pdf
> No errors during installation. [...]

Can you clarify what commands you used? The simplest way to build and install picotool from your picotool build directory (assuming PICO_SDK_PATH environment variable is set) is
```
cmake ..
cmake --build .
sudo cmake --install .
```

There have been multiple fixes to the inclusion in specific circumstances since the 2.0.0 SDK release, so I would recommend using the develop branches of the SDK and picotool if you're still having issues.

all2coolnick, 2024-10-22:
Thanks [will-v-pi](https://github.com/will-v-pi). 
I used the code for the Raspberry Pi 'getting-started-with-pico.pdf' from their web site.
First I cloned the pico-sdk and examples repos ... 
```
$ git clone https://github.com/raspberrypi/pico-sdk.git --branch master
$ cd pico-sdk
$ git submodule update --init
$ cd ..
$ git clone https://github.com/raspberrypi/pico-examples.git --branch master
```

Then for the toolchain ...
```
$ sudo apt update
$ sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi build-essential
```
For picotool ...

```
$ git clone https://github.com/raspberrypi/picotool.git
$ cd picotool
$ mkdir build
$ cd build
$ export PICO_SDK_PATH=~/pico/pico-sdk
$ cmake ../
$ make
```
Lastly, as I am building on MacOS ...
`
$ brew install libusb pkg-config
`

If I delete the whole pico-tool folder, I can build pico-sdk; I just get a warning saying picotool cannot be found and will be downloaded. This happens every time I re-build so it's nt even finding it's own installed version.
Can you clarify what you mean by 'using the develop branches of the SDK and picotool'? Is this just cloning from a different repo?
Thanks

will-v-pi, 2024-10-22:
Firstly, try adding `sudo make install` to the end of the picotool commands, and that should fix it. The SDK can't find the picotool you've built unless you install it (see the picotool readme for more details)

If that doesn't fix it, then to use the develop branches add `--branch develop` to the end of the `git clone` commands - this will clone from the same repo, but use the develop branches rather than the master branches

matsobdev, 2024-10-22:
> Thanks for the reply and sorry to be dense but can you clarify in what context and sequence to use the suggested commands? I’m unfamiliar with -D.

Manually, when building on Ubuntu (20.04) and Windows (7) had problem with SDK finding picotool even with:
```
set(picotool_DIR /home/mateush/pico-sdk-2.0.0/tools/picotool) #location may vary :D
```
inside `CMakeLists.txt`. So according to `picotool` Github page was building it what way:
```
cmake -G "Ninja" -D CMAKE_INSTALL_PREFIX=/home/mateush/pico-sdk-2.0.0/tools/picotool -D PICOTOOL_FLAT_INSTALL=1 -S ~/picotool_source -B ~/picotool_source/build
cmake --build ~/picotool_source/build
```
Location inside `CMAKE_INSTALL_PREFIX` needs to be the same like inside `picotool_DIR`. If not, SDK is unhappy. I just keep SDK binaries inside SDK folder so just manually copied files there. But `picotool` directory need contain files like official binaries [here](https://github.com/raspberrypi/pico-sdk-tools).

will-v-pi, 2024-10-24:
> Location inside `CMAKE_INSTALL_PREFIX` needs to be the same like inside `picotool_DIR`.

Just to clarify, `CMAKE_INSTALL_PREFIX should not be set to the same as `picotool_DIR` - as stated in the picotool [README](https://github.com/raspberrypi/picotool/tree/master?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk) `picotool_DIR` should be set to `CMAKE_INSTALL_PREFIX/picotool` when using custom paths like this. This is likely why you're having problems.

all2coolnick, 2024-10-24:
Thanks all. I will try this.

closed 2024-11-05 by all2coolnick