raspberrypi/pico-sdk #1992: Inconsistent spelling in symbol definition: `interp_add_accumulater`

smitelli, 2024-10-21:
In [hardware/interp.h](https://github.com/raspberrypi/pico-sdk/blob/2.0.0/src/rp2_common/hardware_interp/include/hardware/interp.h#L441), the definition of the inline function `interp_add_accumulater` doesn't match the dictionary spelling of "accumulator" used anywhere else in the project or its documentation.

Since changing it could break existing user code, is this the type of thing where there is precedent to keep the incorrect spelling as an alias?

peterharperuk, 2024-10-21:
Usually we add a correct version of the function but keep the old version around to maintain compatibility

lurch, 2024-10-22:
Yeah the usual approach is to rename the actual function itself, and then add a `#define` for the old name, so that existing code keeps working; similar to [this](https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_i2c/include/hardware/i2c.h#L209) .

So in your case you'd probably want to add something like:
```
#define interp_add_accumulater(interp, lane, val) interp_add_accumulator(interp, lane, val)
```

kilograham, 2024-11-19:
merged into develop

closed 2024-11-19 by kilograham