raspberrypi/pico-sdk #1997: The program gets stuck during I2C communication

Octopus1633, 2024-10-24:
Hello,

I encountered a strange problem. When using RP2350 with 2MB or 4MB FLASH, everything worked fine. When I changed to 8MB or 16MB, I needed to adjust **PICO_FLASH_SPI_CLKDIV** or **PICO_XOSC_STARTUP_DELAY_MULTIPLIER**. After adjustment, 8MB can run normally, but 16MB occasionally fails.

I debugged the program and found that when it failed to run, the program was stuck in a loop, which was called by the program in the **i2c_write_blocking_internal** function:
```c
do {
if (timeout_check) {
timeout = timeout_check(ts, false);
abort |= timeout;
}
tight_loop_contents();
} while (!timeout && !(i2c->hw->raw_intr_stat & I2C_IC_RAW_INTR_STAT_TX_EMPTY_BITS));
```
I tried to print out I2C_IC_RAW_INTR_STAT_TX_EMPTY_BITS and found that the bit was always 0 and did not change. What could be the reason for this?

lurch, 2024-10-24:
Could you provide more details about which exact Flash chips you're using, and which RP2350 board you're using?

Also, if you're able to provide a more complete code-example, that might be useful.

Octopus1633, 2024-10-25:
Thanks for your reply, I have solved it. It seems that PICO_FLASH_SPI_CLKDIV and PICO_XOSC_STARTUP_DELAY_MULTIPLIER are set too large, which may cause I2C to fail to communicate after startup.

closed 2024-10-25 by Octopus1633