raspberrypi/pico-sdk #2007: flash_safe_execute timeout under FreeRTOS

jackwilsdon, 2024-10-30:
There appears to be a bug in `flash_safe_execute` which causes it to timeout on FreeRTOS, no matter how large the enter/exit timeout is set to.

From some debugging, it seems like we're somehow ending up with a flash lockout task with the wrong "other core number" - maybe it's left over from a previous `flash_safe_execute` call?

```
(gdb) info thread
  Id   Target Id                                                          Frame
* 1    Thread 1 "rp2040.core0" (Name: rp2040.core0, state: debug-request) 0x10006b2e in default_enter_safe_zone_timeout_ms (timeout_ms=<optimized out>)
    at pico-sdk/src/rp2_common/pico_flash/flash.c:157
  2    Thread 2 "rp2040.core1" (Name: rp2040.core1, state: debug-request) 0x2000020e in flash_lockout_task (vother_core_num=0x1)
    at pico-sdk/src/rp2_common/pico_flash/flash.c:126
```

It's worth noting that this isn't fully consistent - it seems to depend on how FreeRTOS chooses to schedule the tasks. The following example code reproduces after a short while when dropped into `hello_freertos` from `pico-examples`:

```c
#include <stdio.h>

#include "FreeRTOS.h"
#include "task.h"

#include "pico/flash.h"
#include "pico/stdlib.h"

void vFlash( void * pvArg )
{
}

void vTask( void * pvArg )
{
    for( ; ; )
    {
        const int xResult = flash_safe_execute( vFlash, NULL, 1000 );
        configASSERT( xResult == PICO_OK );
    }
}

int main( void )
{
    stdio_init_all();

    const BaseType_t xResult = xTaskCreate( vTask, "vTask",
                                            configMINIMAL_STACK_SIZE, NULL, 0,
                                            NULL );
    configASSERT( xResult == pdPASS );

    puts( "STARTIN");
    vTaskStartScheduler();

    return 0;
}
```

jackwilsdon, 2024-10-30:
After a bit more debugging, it seems like the issue is that the lockout task is actually running on the wrong core - its affinity indicates it should be running on core 1:
```
(gdb) p ((TCB_t *) pxReadyTasksLists[31].xListEnd.pxNext.pvOwner)->pcTaskName
$29 = "flash lockout\000\000"
(gdb) x/t &((TCB_t *) pxReadyTasksLists[31].xListEnd.pxNext.pvOwner)->uxCoreAffinityMask 
0x20003f04 <ucHeap+13180>:      00000001
```

Making this (somewhat nasty) change fixes the issue for me - not sure if there's a nicer way to do it:

```
diff --git a/src/rp2_common/pico_flash/flash.c b/src/rp2_common/pico_flash/flash.c
index 0d56067..2bd6c0d 100644
--- a/src/rp2_common/pico_flash/flash.c
+++ b/src/rp2_common/pico_flash/flash.c
@@ -117,6 +117,10 @@ static bool use_irq_only(void) {
 #if PICO_FLASH_SAFE_EXECUTE_USE_FREERTOS_SMP
 static void __not_in_flash_func(flash_lockout_task)(__unused void *vother_core_num) {
     uint other_core_num = (uintptr_t)vother_core_num;
+    while (get_core_num() != (other_core_num ^ 1))
+    {
+        // Wait until we're on the right core.
+    }
     while (lockout_state[other_core_num] != FREERTOS_LOCKOUT_LOCKER_WAITING) {
         __wfe(); // we don't bother to try to let lower priority tasks run
     }
```

jackwilsdon, 2024-10-30:
Seems like setting the core affinity at creation also works:

```diff
diff --git a/src/rp2_common/pico_flash/flash.c b/src/rp2_common/pico_flash/flash.c
index 0d56067..a22e8b1 100644
--- a/src/rp2_common/pico_flash/flash.c
+++ b/src/rp2_common/pico_flash/flash.c
@@ -142,16 +142,14 @@ static int default_enter_safe_zone_timeout_ms(__unused uint32_t timeout_ms) {
         // it only prevents the other core from also entering a critical section.
         // Therefore, we must do our own handshake which starts a task on the other core and have it disable interrupts
         uint core_num = get_core_num();
-        // create at low priority
+        // create at low priority on other core
         TaskHandle_t task_handle;
-        if (pdPASS != xTaskCreate(flash_lockout_task, "flash lockout", configMINIMAL_STACK_SIZE, (void *)core_num, 0, &task_handle)) {
+        if (pdPASS != xTaskCreateAffinitySet(flash_lockout_task, "flash lockout", configMINIMAL_STACK_SIZE, (void *)core_num, 0, 1u << (core_num ^ 1), &task_handle)) {
             return PICO_ERROR_INSUFFICIENT_RESOURCES;
         }
         lockout_state[core_num] = FREERTOS_LOCKOUT_LOCKER_WAITING;
         __sev();
-        // bind to other core
-        vTaskCoreAffinitySet(task_handle, 1u << (core_num ^ 1));
-        // and make it super high priority
+        // make it super high priority
         vTaskPrioritySet(task_handle, configMAX_PRIORITIES -1);
         absolute_time_t until = make_timeout_time_ms(timeout_ms);
         while (lockout_state[core_num] != FREERTOS_LOCKOUT_LOCKEE_READY && !time_reached(until)) {
```

peterharperuk, 2024-11-05:
The change seems harmless enough - do you want to make a PR?

I don't fully understand what's going on, as the code looks like it should work. I don't know how realistic your example is. Maybe it should assert that the thread priority is >0 (idle) or else the flash lockout thread can get to run before the call to vTaskCoreAffinitySet? Even with your fix I get a PICO_ERROR_INSUFFICIENT_RESOURCES (-9) error after awhile. For some reason the flash lockout thread is not getting time to delete itself? Adding sleep_ms(1) after the call to flash_safe_execute works around that.

jackwilsdon, 2024-11-05:
I admit my example isn't realistic, but I'm seeing this issue in the case of downloading firmware in blocks and applying them in a `flash_safe_execute` call, which is a bit more realistic.

I think the insufficient resources issue in my example is caused by the idle task not getting a chance to run, although I didn't see that on my device.

Happy to open a PR for this :+1:

peterharperuk, 2024-11-05:
>  I'm seeing this issue in the case of downloading firmware in blocks

Does the thread calling flash_safe_execute have a priority of zero? It seems odd to me that it would get to run before the call to vTaskCoreAffinitySet.

jackwilsdon, 2024-11-05:
Yes it does, upping the priority did also fix the timeout issue from what I remember, so it got pre-empted by the lockout task.

lurch, 2024-11-21:
Fixed in `develop`

PR #2021 (2024-11-21): Fix flash_safe_execute timeout under FreeRTOS
In some cases, the flash lockout task can start on the wrong core and not have time to move to the correct core before exit is called. This causes a timeout as the exit function is looking at the wrong core when checking for the lockout task.

Fixes #2007.
closed 2024-11-21 by lurch