raspberrypi/pico-sdk #2008: Cloned project not build for pico_w: pico/cyw43_arch.h: No such file or directory

ivanvaccari, 2024-10-30:
I'm having a bit of annoying time building a pico_w project just cloned from github.

The project builds correctly on a "DESKTOP" PC,, but it's not building by defaut on a "LAPTOP" pc where i just cloned the project:

`fatal error: pico/cyw43_arch.h: No such file or directory`

This is my CMakeList.txt:

```
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NEVER EDIT THE NEXT LINES for Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.0.0)
set(toolchainVersion 13_2_Rel1)
set(picotoolVersion 2.0.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico_w CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(chicken_door C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_executable(chicken_door 
    main.cpp 
    DRV8874Controller.cpp 
    DRV8874Controller.h
    LightSensor.cpp
    LightSensor.h
    StatusLed.cpp
    StatusLed.h 
    PWMStatusLed.cpp
    PWMStatusLed.h 
    libs/GpioInterrupts/GpioInterrupts.cpp 
    libs/GpioInterrupts/GpioInterrupts.h 
    libs/PWMOut/PWMOut.cpp
    libs/PWMOut/PWMOut.h 
    libs/WifiStation/WifiStation.cpp
    libs/WifiStation/WifiStation.h
    libs/Timer/Timer.cpp
    libs/Timer/Timer.h
    libs/mqtt/Mqtt.h
    libs/mqtt/Mqtt.cpp
    libs/DnsResolver/DnsResolver.h
    libs/DnsResolver/DnsResolver.cpp
    libs/AdcPin/AdcPin.cpp
    libs/AdcPin/AdcPin.h
    libs/Json/Json.h
    libs/Json/Json.cpp
    libs/Json/tiny-json/tiny-json.h
    libs/Json/tiny-json/tiny-json.c
    libs/HAMqttAutodiscovery/HAMqttAutodiscovery.h
    libs/HAMqttAutodiscovery/HAMqttAutodiscovery.cpp
    libs/Watchdog/Watchdog.h
    libs/Watchdog/Watchdog.cpp
    libs/Rtc/Rtc.h
    libs/Rtc/Rtc.cpp

    )

pico_set_program_name(chicken_door "chicken_door")
pico_set_program_version(chicken_door "0.1")

# Modify the below lines to enable/disable output over UART/USB
pico_enable_stdio_uart(chicken_door 0)
pico_enable_stdio_usb(chicken_door 1)

# Add the standard include files to the build
target_include_directories(chicken_door PRIVATE
  ${CMAKE_CURRENT_LIST_DIR}
  ${CMAKE_CURRENT_LIST_DIR}/.. # for our common lwipopts or any other standard includes, if required
)

# Add any user requested libraries
target_link_libraries(chicken_door 
    pico_stdlib
    pico_cyw43_arch_lwip_threadsafe_background 
    hardware_pwm
    pico_lwip_mqtt
    hardware_adc
    hardware_rtc
)

pico_add_extra_outputs(chicken_door)

```

compile_commands.json **do** include `-DPICO_BOARD=\\\"pico_w\\\"` flag.

I genuinely expected to never bother again about pico_w libs once configured.

Is there some step i'm missing?

ivanvaccari, 2024-10-30:
Turned out pico-vscode extension produced an inconsistent .pico-sdk directory where a lot of files were missing.

That extension apparently does not manage very well not having python3. IT start downoading some files, then breaks somewhere:

[image]

It should check beforehand absence of python and should not write any file in case

closed 2024-10-30 by ivanvaccari

lurch, 2024-10-30:
@will-v-pi Any ideas?

will-v-pi, 2024-10-31:
The python handling has been updated for pico-vscode, so it it wasn't working correctly then try the latest build and it should work better?

ivanvaccari, 2024-11-01:
I installed the extension just before cloning the project so i suppose it was already at the latest version.

will-v-pi, 2024-11-01:
The latest version (0.17.0) was only released yesterday, so you wouldn't have been on the latest version when you initially cloned the project

yoyojacky, 2025-03-14:
I have met  the same problem. 
Setup according to the instructions on my raspberry pi 5:
test code: 
```c 
/**
 * Copyright (c) 2022 Raspberry Pi (Trading) Ltd.
 *
 * SPDX-License-Identifier: BSD-3-Clause
*/

#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"

int main() {
    stdio_init_all();
    if (cyw43_arch_init()) {
        printf("Wi-Fi init failed");
        return -1;
    }
    while (true) {
        cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, 1);
        sleep_ms(250);
        cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, 0);
        sleep_ms(250);
    }
}
```
and 

[image]

[image]

[image]
```CMakeLists.txt 
$ cat CMakeLists.txt
cmake_minimum_required(VERSION 3.13)

include(pico_sdk_import.cmake)

project(test_project C CXX ASM)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

pico_sdk_init()

add_executable(test
        test.c
)

pico_enable_stdio_usb(test 1)
pico_enable_stdio_uart(test 1)

pico_add_extra_outputs(test)

target_link_libraries(test pico_stdlib)
```
cmake is fine:

[image]

and when i build it. 

[image]
so i try to find this header file:

[image]

there is no such a file named "cyw43_arch.h" so what should i do ?

yoyojacky, 2025-03-14:
so i change the header file path from pico/cyw43_arch.h to '../pico-sdk/lib/cyw43-driver/src/cyw43.h' and save it , recompile it.

[image]

[image]

still no luck ,please help.

lurch, 2025-03-14:
Looks like you forgot to add `pico_cyw43_arch_none` to your `target_link_libraries` - see https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/blink/CMakeLists.txt#L4C1-L4C22

will-v-pi, 2025-03-14:
Also, your `PICO_BOARD` is `pico` not `pico_w` - you need to run cmake with `-DPICO_BOARD=pico_w`