raspberrypi/pico-sdk #2010: C++ std::format crashes with numbers

Slion, 2024-11-02:
SDK 2.0.0
GCC 13.2.1

`std::format("Hello {}!\n", "World")` works
`std::format("Hello {}!\n", 32)` crashes

Slion, 2024-11-02:
@vitaut Any idea what could be causing this?

Slion, 2024-11-02:
Same issue with `fmt::format("Hello {}!\n", 32);` using libfmt v11.0.2 that's not specific libstdc++ apparently.
This is targeting ARM btw in case it was not obvious.

vitaut, 2024-11-02:
It's unlikely that both libstdc++ and {fmt} are broken in the same way so the issue is probably in your toolchain.

Slion, 2024-11-02:
Yeah, that's why I opened this issue here. Let's see if the guys at Raspberry Pi can help.

Slion, 2024-11-02:
Added the callstack to first post.

Slion, 2024-11-02:
Looks like it does try to deal with the int but the number of arguments is obviously corrupted:
[image]

Slion, 2024-11-03:
As a side note std::format need 157kB of flash where fmt::format takes up 61kB.

geurtv, 2024-11-03:
Works for me ...

Tried it with [GCC 13.3.Rel1](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) for both RP2040 and RP2350 and didn't see any issues.

Can you provide a full code example (with build options)?

Slion, 2024-11-03:
> Works for me ...
> 
> Tried it with [GCC 13.3.Rel1](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) for both RP2040 and RP2350 and didn't see any issues.
> 
> Can you provide a full code example (with build options)?

That's weird, thanks for trying it out.
You are not using the same toolchain though.
Were you building from Windows or Linux?

kilograham, 2024-11-03:
It works fine for me with `arm-none-eabi-gcc (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.1 20231009`

Are you maybe overflowing your stack? Actual repro source would help

Slion, 2024-11-03:
> Are you maybe overflowing your stack?

I don't think I'm, otherwise text formatting would be the least of my problems.
I've got btstack, TinyUSB, lwip httpd, pico-vfs and LCD display all running fine, mostly 😁
I have `PICO_USE_STACK_GUARDS` enabled.
My heap it pretty full 78kB from 96kB available but still ok I guess.

> Actual repro source would help

It's a closed source repo and I have not tried to isolated the issue yet. It was based on the btstack HID host example so the core architecture is still the same, currently using poll async context.

```
set(CMAKE_C_STANDARD 17)
set(CMAKE_CXX_STANDARD 20)
set(PICO_CXX_ENABLE_EXCEPTIONS 0)
set(PICO_CXX_ENABLE_RTTI 0)
```

Though I noticed something funny in my callstack it says `c++11eabi` and while compiling I get `note: parameter passing for argument of type 'std::format_args' {aka 'std::basic_format_args<std::basic_format_context<std::__format::_Sink_iter<char>, char> >'} changed in GCC 9.1`.

Slion, 2024-11-03:
I can't reproduce it myself anymore. It just works for some reason. I did a bunch a refactoring since yesterday but nothing that should affect that really. Sorry I wasted your time. Thanks for the help.

Slion, 2024-11-24:
Now it's happening again…

kilograham, 2024-11-24:
i wonder if your stack is misaligned? (the formatting may require 8 byte alignment)

Slion, 2024-11-24:
> i wonder if your stack is misaligned? (the formatting may require 8 byte alignment)

How do I fix that?

kilograham, 2025-01-06:
probably first set a breakpoint in the debugger and check whether SP is 8 byte aligned - if it isn't, you need to figure out why (again debugger would help here, as you can look at each stack frame)

kilograham, 2026-09-30:
Closing as we haven't heard back.

closed 2026-09-30 by kilograham