raspberrypi/pico-sdk #2011: devlopment branch: incompatibility between HID host and repeating timer

UKTailwind, 2024-11-04:
Code works on V2.0.0 but was advised to try development branch to fix an issue with a repeating timer occasionally stopping after an extended period

`    add_repeating_timer_us(-1000, timer_callback, NULL, &timer);`

The development branch appear to correct this except if tuh_task(); is called in which case, if a hid device is plugged in, the timer stops.
In the example output below I call tuh_task() for the first time 10 seconds after the program starts. The time is updated by the timer callback. You can see in the output after the call to tuh_host() the timer stops updating. Then after an extended period (c1min) the hid device spontaneously disconnects and reconnects and the timer is running again. During all of this the main program is working as expected.
[image]

lurch, 2024-11-04:
ping @hathach in case he has any ideas?

peterharperuk, 2024-11-04:
"The time is updated by the timer callback." Sorry, what do you mean by this exactly?

UKTailwind, 2024-11-04:
The time is maintained in the timer callback using the usual simple code. The fact this it isn't updating shows the callback isn't executing. This is just a convenient way to demonstrate it.  I've also shown it with more direct monitoring by comparing in the main code a counter incremented in the callback with time_us_64()
`    if(++SecondsTimer >= 1000) {
        SecondsTimer -= 1000; 
    #ifndef PICOMITEWEB
        if(ExtCurrentConfig[PinDef[HEARTBEATpin].pin]==EXT_HEARTBEAT)gpio_xor_mask64(1<<PinDef[HEARTBEATpin].GPno);
    #endif
            // keep track of the time and date
        if(++second >= 60) {
            second = 0 ;
            if(++minute >= 60) {
                minute = 0;
                if(++hour >= 24) {
                    hour = 0;
                    if(++day > DaysInMonth[month] + ((month == 2 && (year % 4) == 0)?1:0)) {
                        day = 1;
                        if(++month > 12) {
                            month = 1;
                            year++;
                        }
                    }
                }
            }
        }
    }

`

peterharperuk, 2024-11-05:
Is your code in github somewhere? It sounds like you might be calling tuh_task inside of the timer call back which makes me wonder how you avoid calling sleep inside of an exception handler?

peterharperuk, 2024-11-12:
Closing this. I tried to reproduce the problem without luck. Reopen if you can point me towards some code that reproduces the problem.

closed 2024-11-12 by peterharperuk