raspberrypi/pico-sdk #2012: DHCSR register missing from SVD in RP2040

0x3333, 2024-11-04:
I was looking for the `DHCSR` register, but couldn't find it. There is a mention to it in the `AIRCR` register, but nothing else.

In the RP2350 it is defined.

The problem is that it doesn't get generated int the CMSIS headers.

Am I missing something here? 

Much appreciate.

lurch, 2024-11-04:
ping @matiasilva

0x3333, 2024-11-04:
> I don't believe the registers are accessible from software on RP2040 (M0+)

Sounds right, because reading it result in 0x00000000. 

So, no way to know if the debugger is attached?

Wren6991, 2024-11-12:
> So, no way to know if the debugger is attached?

There is no standard Arm way to do this as far as I know.

However you can check the status of the CDBGPWRUPREQ bit from the DP CTRL/STAT register by reading this POWMAN register:

```
// =============================================================================
// Register    : POWMAN_CURRENT_PWRUP_REQ
// Description : Indicates current powerup request state
//               pwrup events can be cleared by removing the enable from the
//               pwrup register. The alarm pwrup req can be cleared by clearing
//               timer.alarm_enab
//               0 = chip reset, for the source of the last reset see
//               POWMAN_CHIP_RESET
//               1 = pwrup0
//               2 = pwrup1
//               3 = pwrup2
//               4 = pwrup3
//               5 = coresight_pwrup
//               6 = alarm_pwrup
#define POWMAN_CURRENT_PWRUP_REQ_OFFSET _u(0x0000009c)
#define POWMAN_CURRENT_PWRUP_REQ_BITS   _u(0x0000007f)
#define POWMAN_CURRENT_PWRUP_REQ_RESET  _u(0x00000000)
#define POWMAN_CURRENT_PWRUP_REQ_MSB    _u(6)
#define POWMAN_CURRENT_PWRUP_REQ_LSB    _u(0)
#define POWMAN_CURRENT_PWRUP_REQ_ACCESS "RO"
// =============================================================================
```

Wren6991, 2024-11-12:
You can also check if a processor is halted for debug using this SYSCFG register:

```
// =============================================================================
// Register    : SYSCFG_PROC_CONFIG
// Description : Configuration for processors
#define SYSCFG_PROC_CONFIG_OFFSET _u(0x00000000)
#define SYSCFG_PROC_CONFIG_BITS   _u(0x00000003)
#define SYSCFG_PROC_CONFIG_RESET  _u(0x00000000)
// -----------------------------------------------------------------------------
// Field       : SYSCFG_PROC_CONFIG_PROC1_HALTED
// Description : Indication that proc1 has halted
#define SYSCFG_PROC_CONFIG_PROC1_HALTED_RESET  _u(0x0)
#define SYSCFG_PROC_CONFIG_PROC1_HALTED_BITS   _u(0x00000002)
#define SYSCFG_PROC_CONFIG_PROC1_HALTED_MSB    _u(1)
#define SYSCFG_PROC_CONFIG_PROC1_HALTED_LSB    _u(1)
#define SYSCFG_PROC_CONFIG_PROC1_HALTED_ACCESS "RO"
// -----------------------------------------------------------------------------
// Field       : SYSCFG_PROC_CONFIG_PROC0_HALTED
// Description : Indication that proc0 has halted
#define SYSCFG_PROC_CONFIG_PROC0_HALTED_RESET  _u(0x0)
#define SYSCFG_PROC_CONFIG_PROC0_HALTED_BITS   _u(0x00000001)
#define SYSCFG_PROC_CONFIG_PROC0_HALTED_MSB    _u(0)
#define SYSCFG_PROC_CONFIG_PROC0_HALTED_LSB    _u(0)
#define SYSCFG_PROC_CONFIG_PROC0_HALTED_ACCESS "RO"
// =============================================================================
```

However this is less useful for software running on-device, since your code is by definition not running when you are halted. (It's handy for debugging multi-core debug translators)

kilograham, 2024-11-13:
closing this, as i don't think we want to add DHCSR to the RP2040 SVD - feel free to complain if you do

closed 2024-11-13 by kilograham