raspberrypi/pico-sdk #2014: Reported system clock rate incorrect when configuring pll settings in cmakelists

CarlAndersson, 2024-11-04:
I need some specific system clock rates for my audio-related project with an external ADC. Found the `vcocalc.py` script after some quick searches. It shows the option to configure the system pll from cmake instead of in the c code, e.g.
```
target_compile_definitions(audio PRIVATE
        PLL_SYS_REFDIV=1
        PLL_SYS_VCO_FREQ_HZ=1476000000
        PLL_SYS_POSTDIV1=5
        PLL_SYS_POSTDIV2=2
)
```
This seems to work as intended, checking the system clock rate with a frequency counter gave me 147.6 MHz, as expected.
However, the value returned by `clock_get_hz` is still 150 MHz (pico 2).
If I also set `SYS_CLK_HZ` in cmake, `clock_get_hz` will return this value, but it will return whatever value I put into cmake.

Looking into the code I think this gets set [here](https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/pico_runtime_init/runtime_init_clocks.c#L90), which in turn simply uses the clock speed defined [here](https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2350/hardware_regs/include/hardware/platform_defs.h#L99).

If this is intended behaviour, perhaps the `vcocalc.py` script should include a line on also setting `SYS_CLK_HZ` in cmake?

[code.zip](https://github.com/user-attachments/files/17620979/pico-sdk-bug-report.zip)

<details><summary>Test code and output</summary>

```c
uint32_t reported_before_setting = clock_get_hz(clk_sys);
uint32_t counted_before_setting = frequency_count_khz(CLOCKS_FC0_SRC_VALUE_CLK_SYS);
uint32_t freq_hz = 123 * MHZ;  // 123 MHz can give us close to the bitrate we need.
set_sys_clock_hz(freq_hz, true);
uint32_t reported_after_setting = clock_get_hz(clk_sys);
uint32_t counted_after_setting = frequency_count_khz(CLOCKS_FC0_SRC_VALUE_CLK_SYS);
stdio_init_all();

puts("\n\n==== Pico is booted ====");

printf("Reported before: %ld, reported after %ld\n", reported_before_setting, reported_after_setting);
printf("Counted before: %ld, counted after %ld\n", counted_before_setting, counted_after_setting);

```
```
==== Pico is booted ====
Reported before: 137600000, reported after 123000000
Counted before: 147600, counted after 123000
```
</details>

peterharperuk, 2024-11-05:
cc @andygpz11

andygpz11, 2024-11-05:
@CarlAndersson Thank you for your report and details.  

We agree, it appears that `SYS_CLK_HZ` does need to be defined to correct the reporting.

I have just raised [PR2022](https://github.com/raspberrypi/pico-sdk/pull/2022)

kilograham, 2024-11-06:
merged into develop

PR #2022 (2024-11-06): Add the required `SYS_CLK_HZ` definition to the suggested cmake output
Fixes #2014
commit 6587f5cc (2024-11-06): Add the required `SYS_CLK_HZ` definition to the suggested cmake output (#2022)

Fixes #2014
closed 2024-11-06 by kilograham