raspberrypi/pico-sdk #2024: Method to see intermediate sha256 result

LinuxJedi, 2024-11-06:
I'm working on porting wolfSSL's wolfCrypt to use the RP2350 hardware extensions. I'm a little stuck implementing `wc_Sha256GetHash`.

This function needs to get the intermediate SHA256 result, half way through calculation, before Final. It is used, amongst other things, in our TLSv1.3 implementation.

I've tried digging into the SDK to figure a way to do this, `sha256_get_result()` being an obvious candidate. But it appears you need to pump padding in first. Which likely makes the final result invalid when you do more updates.

Is there a way to do this that I'm not seeing? Can this be something added to the SDK?

peterharperuk, 2024-11-06:
> But it appears you need to pump padding in first. 

I think the hardware will only give you a valid checksum if it has been given a multiple of 64 bytes of data (no padding here?). It's awhile since I looked at what the hardware does so I don't know what sha256_get_result returns when it thinks the checksum is invalid.

LinuxJedi, 2024-11-06:
> I think the hardware will only give you a valid checksum if it has been given a multiple of 64 bytes of data (no padding here?). It's awhile since I looked at what the hardware does so I don't know [...]

In an invalid state it returns `0x6a09e667bb67ae853c6ef372a54ff53a510e527f9b05688c1f83d9ab5be0cd19`. `write_padding()` does the normal SHA padding thing to bump a smaller message on Final to 64 bytes so that you can get a valid result.

I think the workaround might be to manipulate the byte counter in the state after `write_padding()`, but I'm going to need to play with it a bit to figure it out.

peterharperuk, 2024-11-06:
I'd be surprised if it worked if you started adding padding half way through a message? Or maybe I've misunderstood what you're suggesting.

LinuxJedi, 2024-11-06:
> I'd be surprised if it worked if you started adding padding half way through a message? Or maybe I've misunderstood what you're suggesting.

Yea, it was a dumb idea of mine. I don't think this is possible given the constraints of the hardware. I might have to just implement to the RNG acceleration and skip the SHA256. I'll close for now as that would make it out-of-scope for the SDK.

closed 2024-11-06 by LinuxJedi