raspberrypi/pico-sdk #2030: PIO gpiobase not set with pin >= 32

sfe-SparkFro, 2024-11-08:
I tried running the [PIO WS2812 example](https://github.com/raspberrypi/pico-examples/blob/master/pio/ws2812/ws2812.c) with an RP2350B, and found that the LED does not work when using a GPIO pin >=32. After some digging, it appears the root problem is that the `pio->gpiobase` is not getting set to `16`. If I manually set the value, it works fine with GPIO pins >= 32.

Where is the correct place for `gpiobase` to be set? I put the following check into [`pio_sm_set_consecutive_pindirs()`](https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/hardware_pio/pio.c#L270) which worked for me, but I'm guessing it may be better to go somewhere else.

```
if(pin > 31 && pio_get_gpio_base(pio) == 0)
    pio->gpiobase = 0x10;
```

sfe-SparkFro, 2024-11-08:
Also, not sure whether #1834 is related

lurch, 2024-11-11:
I wonder if this ought to be classed as a bug in the PIO WS2812 example? Perhaps there needs to be a call to `pio_set_gpio_base()` in https://github.com/raspberrypi/pico-examples/blob/develop/pio/ws2812/ws2812.pio#L36 ? :shrug: 
Disclaimer: I've not written any PIO code myself :joy:

sfe-SparkFro, 2024-11-13:
I did consider whether it should be an issue with the example, but decided against that because IMO the SDK should abstract away things like setting `gpiobase`. The user should only have to worry about which PIO and pin they're using, which is why I opened the issue here. Although I've not written much PIO code either, so I'm not sure if this is something the user should actually need to worry about, but my naïve feeling is that the user shouldn't have to do things differently for GPIO >=32.

lurch, 2024-11-13:
From what little I understand, I'm not sure how much the SDK can do "automatically" because you're **never** going to have PIO code which simultaneously works with pins 10 and 40? (because one is in the 0-32 range and the other is in the 16-48 range)

StijnKuipers, 2024-11-14:
also - it is unclear how other pio related functions behave after a base is set. 

 pio_set_gpio_base(pio, 16);
 pio_gpio_init(pio, somepin);    <-- does somepin now need to be offset by 16? or is gpio number absolute? 
 sm_config_set_out_pins(&sm, somepin, 1); <-- how about here? 

 same for all other pio and sm functions that expect a pin number

shtirlic, 2024-11-14:
Here some info about the sm functions, also there is some magic with pioasm detection what range of pins is used, affecting fail to set gpio base

shtirlic, 2024-11-14:
> From what little I understand, I'm not sure how much the SDK can do "automatically" because you're **never** going to have PIO code which simultaneously works with pins 10 and 40? (because one is in the 0-32 range and the other is in the 16-48 range)
 Theoretically it can be done if you can merge 2 PIO blocks , for Example PIO1+PIO2 and  we get a virtual PIO3 with all 48 pins and base 0

peterharperuk, 2024-11-14:
@StijnKuipers I believe both the functions you mention there handle things properly, i.e. you can pass a gpio number >31 and they should do the right thing.

Octopus1633, 2024-11-19:
@StijnKuipers @sfe-SparkFro 
Hello, I would like to ask how you solved this problem.I am testing the [hello_pio](https://github.com/raspberrypi/pico-examples/tree/master/pio/hello_pio) program. And I also use the RP2350B. I want to use pin 45 to output high and low levels alternately.

I tried adding

```C
pio_set_gpio_base(pio,16);
```
and changing the pin to 45

```C
#define HELLO_PIO_LED_PIN 45
```
and found that the pin did not respond.

Then I tried to change the pin to a relative value

```C
#define HELLO_PIO_LED_PIN (45-16)
```
Still no response, why is this?

sfe-SparkFro, 2024-11-19:
@Octopus1633 See my first message in this thread. That solution worked for my specific situation with the WS2812 PIO example, but I'm not sure whether that will work for every situation. I don't know enough about the PIO API to know what the "right" solution would be.

This issue has been assigned to the [v2.1.0 milestone](https://github.com/raspberrypi/pico-sdk/milestone/22), so I'm guessing it will be addressed soon, but they're probably finishing work on other things right now.

peterharperuk, 2024-11-20:
Closing this as I'm fairly happy there's nothing else to fix here. Please reopen if you disagree. The simple answer is to use pio_claim_free_sm_and_add_program_for_gpio_range.

closed 2024-11-20 by peterharperuk

poetaster, 2026-02-12:
> > but if I attempt to create the next object (doesn't matter if I set another PIO) it's a hard crash
> 
> You're not giving us a lot to go on. Do you have a call stack? Are you checking "success" is true?

success is always true, until you set PIO and base.  If you switch base, you have to switch PIO it seems. But, no can do. I'll try to produce a pure sdk example.

poetaster, 2026-02-13:
> The gpio base applies to everything on the pio instance. So if you need a different gpio base you have to use a different pio. pio_claim_free_sm_and_add_program_for_gpio_range tries to choose a pio that is compatible with the gpio you are asking for, or else it will choose another one

Yes, I'm aware of that. The point is that in a  program that attempts to use both ranges, I generally get a hung processor. Given my above class, if instantiate and 'begin':
```
PioEncoder encB(39, PIO pio1);
PioEncoder encC(32, PIO, pio1);
PioEncoder encD(36, PIO, pio1);

PioEncoder encA(6, PIO, pio0);
````
followed 
```
pio_set_gpio_base(PIO pio0, 16);
encB.begin();
encC.begin();
encD.begin();
pio_set_gpio_base(PIO pio0, 16);
encA.begin();
````
This succeeds BUT as soon as I call encA, it's a  hard crash 
```
  Serial.print(encA.getCount());
  Serial.print("\t\t");
  
  Serial.print(encB.getCount());
  Serial.print("\t\t");
  
  Serial.print(encC.getCount());
  Serial.print("\t\t");

  Serial.print(encD.getCount());
```

If I don't use encA, the program runs.

poetaster, 2026-02-13:
Which, I know, simply shows that the pio0 was not allocated. Question is, should it? In the above case, it's only 4 sms on 8 pins, 3 in the > 31 range and one below.

lurch, 2026-02-14:
> If I understand correctly, SM, OFFSET and obviously PIO need to be tracked separately for lower and upper base, correct?

Each PIO block (there's 2 in RP2040 and 3 in RP2350) has 4 state machines and 32 words of instruction memory. The "offset" records where in the PIO's instruction-memory each PIO-program starts (if you're clever, you can squeeze several different PIO programs into those 32 words). Each state machine in a PIO block can either run a different PIO program, or run the same PIO program. So in the "simple" case where you've only got a single PIO program and multiple state machines running the same program on different pins, I guess you could just force the offset to always be 0.
On RP2040 and RP2350A, there are only 30 GPIO pins, so PIO always addresses the "lower" range. On RP2350B, each PIO can either address the "lower" range (GPIOs 0-31) or the "upper" range (GPIOs 16-47) and this applies to all state machines in that PIO block.

I dunno if it helps, but I've tried to make the `pio_blink` example a bit easier to understand: https://github.com/raspberrypi/pico-examples/pull/737

lurch, 2026-02-14:
I think the only global state you need to keep track of is which PIOs are using the "low range" and which PIOs are using the "high range"? I guess if you want to keep things simple, one approach might be to have a "low-range encoder" which only allows GPIOs 0-31, and a "high-range encoder" which only allows GPIOs 16-47 ?