raspberrypi/pico-sdk #2037: Top-level README.md instructs use of wrong version of CMake

IJeffray, 2024-11-13:
In current master, cmake/generic_board.cmake uses a POP_FRONT command which only exists in CMake >= 3.15
But the README.md has examples showing cmake_minimum_required(VERSION 3.13)
This leads to significant confusion when the build just spins with the repeated error "list does not recognize sub-command POP_FRON"

CMake 3.13 is the latest in Debian 10, which is admittedly old in the tooth now, but no amount of "apt upgrade" will resolve it.

But there's little point in having a 'minimum required' version line with demonstrably the wrong version in it.
Correcting this would mean that a suitable sane error is thown... "CMake 3.15 or higher is required.  You are running version 3.13.4"

lurch, 2024-11-13:
See the discussion in #1879

lurch, 2024-11-13:
@IJeffray Are you using the `master` or `develop` branch of pico-sdk?

IJeffray, 2024-11-13:
@lurch It's the same on both

lurch, 2024-11-13:
@IJeffray Hmmm, #1879 has been merged into the `develop` branch, so if you're using that branch you _should_ get an error message from CMake, telling you that you need a newer version.

BTW, I assume you're aware that Debian 10 has already reached end of LTS support on 30th June 2024?

IJeffray, 2024-11-13:
@lurch Yeah, I'm aware Debian 10's considered old hat, but it still doesn't really change the underlying issue that things are out of sync.
Sorry I somehow didn't see #1879 before reporting this.
https://github.com/raspberrypi/pico-sdk/tree/develop still shows 3.13 in the README.md though - so whatever was changed for #1879 didn't change that.

lurch, 2024-11-13:
> https://github.com/raspberrypi/pico-sdk/tree/develop still shows 3.13 in the README.md though - so whatever was changed for #1879 didn't change that.

Yes, as discussed in #1879 the "top-level" CMake files only need CMake 3.13, so that's why the README.md says that too. You might be using a configuration which means that your particular build doesn't need `cmake/generic_board.cmake`, in which case CMake 3.13 would still be sufficient.

IJeffray, 2024-11-13:
I see. Confusing, but also makes sense I guess.  Sorry for wasting time as it's already fixed for the next release.

closed 2024-11-13 by IJeffray