raspberrypi/pico-sdk #2047: I would say, that `CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_LPOSC_CLKSRC` definition is missing

matsobdev, 2024-11-16:
...inside https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2350/hardware_regs/include/hardware/regs/clocks.h, according to https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_hardware_clocks. It seems to me, but I'm not sure, tho.

lurch, 2024-11-16:
A bit more fully-fleshed info, after a bit of digging...
I think the complaint is that this table in the documentation:
[image]
which is in the RP2350 section of https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#detailed-description-4 says that `clk_ref` can use the Aux Sources "GPIO in 0" (`CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_GPIN0`), "GPIO in 1" (`CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_GPIN1`), "USB PLL" (`CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB`) or "LPOSC" (`CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_LPOSC_CLKSRC`). However https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2350/hardware_regs/include/hardware/regs/clocks.h#L543 only lists the values for `CLOCKS_CLK_REF_CTRL_AUXSRC` of `CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB`, `CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_GPIN0`, `CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_GPIN1` and `CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB_PRIMARY_REF_OPCG` and therefore appears to be "missing" `CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_LPOSC_CLKSRC`.

However, if we look at the `CLOCKS: CLK_REF_CTRL` register on page 532 of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf we can see that it also only lists the possible values of `AUXSRC` as being `CLKSRC_PLL_USB`, `CLKSRC_GPIN0`, `CLKSRC_GPIN1` and `CLKSRC_PLL_USB_PRIMARY_REF_OPCG`. (Also, as `AUXSRC` is only a 2-bit field, it can't have more than 4 different values.)

The table in the HTML documentation linked above is ultimately derived from https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/hardware_clocks/include/hardware/clocks.h#L105 so I guess either the register descriptions for the `CLOCKS: CLK_REF_CTRL.AUXSRC` fields are mis-labeled, or that Doxygen table in `clocks.h` is wrong? :thinking: 
ping @liamfraser

lurch, 2024-11-16:
P.S. I didn't know what OPCG was, but after a bit of searching it seems to be "On-Product Clock Generator" and it seems to be related purely to the factory-testing of chips. https://www.youtube.com/watch?v=XUJVGjDfJJg (ATE = Automated Test Equipment)

matsobdev, 2024-11-17:
Problem was, when I was trying to run chip from LPOSC by:
```
clock_configure(clk_ref, CLOCKS_CLK_REF_CTRL_SRC_VALUE_LPOSC_CLKSRC, 0, 32768, 32768);
clock_configure(clk_sys, CLOCKS_CLK_SYS_CTRL_SRC_VALUE_CLK_REF, 0, 32768, 32768);
```
and it just won't deliver. Turning LED right after that and it stays like that, but should blink in a loop. Adding some sleep, then turning LED - nothing happens. Then I turned to auxiliary but it's not there. Now using:
```
clock_configure(clk_ref, CLOCKS_CLK_REF_CTRL_SRC_VALUE_CLKSRC_CLK_REF_AUX, CLOCKS_CLK_REF_CTRL_AUXSRC_VALUE_CLKSRC_PLL_USB_PRIMARY_REF_OPCG, 32768, 32768);
```
instead is not working.

liamfraser, 2024-11-19:
So yes I think as Andrew said this is an issue with the table in the documentation

liamfraser, 2024-11-19:
I will take a quick look at this clock configuration but it will be extremely slow. By changing clk_sys you will be slowing down the flash to clock to 16KHz with the default divisor of 2. You are running the device 3814 times slower than 125MHz. What code are you using to flash the LED? If you have changed clk_ref without reconfguring the ticks then the timer will be 366 times slower than normal (12mhz / 32768)

kilograham, 2024-11-19:
merged into develop

closed 2024-11-19 by kilograham

matsobdev, 2024-11-19:
```
#include "pico/stdlib.h"
#include "hardware/clocks.h"

void main() {
	gpio_init(25);
	gpio_set_dir(25, GPIO_OUT);
	
	clock_configure(clk_ref, CLOCKS_CLK_REF_CTRL_SRC_VALUE_LPOSC_CLKSRC, 0, 32768, 32768);
	clock_configure(clk_sys, CLOCKS_CLK_SYS_CTRL_SRC_VALUE_CLK_REF, 0, 32768, 32768);
	
	while (true) {
		gpio_put(25, 1);
		sleep_ms(1);
		gpio_put(25, 0);
		sleep_ms(1);
	}
}
```

lurch, 2024-11-19:
> So as you said, further configuration is required.

In his previous comment, Liam mentioned "If you have changed clk_ref without reconfguring the ticks then the timer will be 366 times slower than normal".
See section 8.5 of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf for more information about that.

matsobdev, 2024-11-19:
Thanks. I was there before, but didn't connect the dots so far :D

liamfraser, 2024-11-20:
The timer is designed to run from a 1us tick, which is 1 tick every 1MHz. The default configuration with the XOSC running at 12MHz counts to 12 to generate each 1us/1mhz tick. The 32KHz clock is too slow to provide a 1us tick. So I'd be tempted to add a scaling factor of 366.21 to your sleep times if you want them to be accurate.