raspberrypi/pico-sdk #2051: Pins greater than 32 use PIO functions

Octopus1633, 2024-11-19:
Hello, I am testing the [hello_pio](https://github.com/raspberrypi/pico-examples/tree/master/pio/hello_pio) program. And I am using the RP2350B development board. I want to use pin 45 to output high and low levels alternately.

I tried adding
```c
pio_set_gpio_base(pio,16);
```
and changing the pin to 45
```c
#define HELLO_PIO_LED_PIN 45
```

and found that the pin did not respond.

Then I tried to change the pin to a relative value
```c
#define HELLO_PIO_LED_PIN (45-16)
```

Still no response, why is this?

peterharperuk, 2024-11-19:
Duplicate https://github.com/raspberrypi/pico-sdk/issues/2030

Octopus1633, 2024-11-19:
Hi, I have previously looked at #2030 and modified the source code, but I still can't get GPIO45 to work properly.

Octopus1633, 2024-11-19:
> Duplicate #2030

Hi, I have previously looked at https://github.com/raspberrypi/pico-sdk/issues/2030 and modified the source code, but I still can't get GPIO45 to work properly.

peterharperuk, 2024-11-19:
Sorry, misread your post. I'll check if we're missing something else.

peterharperuk, 2024-11-19:
@Octopus1633  I've tested this and it works fine. You do NOT need to take 16 off the pin when running the hello_pio program. You don't even have to call pio_set_gpio_base as the call to pio_claim_free_sm_and_add_program_for_gpio_range does that for you.

The following pio function are called in this program and they all correctly handle pins >= 32
sm_config_set_out_pins
pio_gpio_init
pio_sm_set_consecutive_pindirs

Just make sure you are testing on a board that does NOT set PICO_RP2350A 1

closed 2024-11-19 by peterharperuk

Octopus1633, 2024-11-20:
> @Octopus1633 I've tested this and it works fine. You do NOT need to take 16 off the pin when running the hello_pio program. You don't even have to call pio_set_gpio_base as the call to pio_claim_free_sm_and_add_program_for_gpio_range does that for you.

Thank you for your reply. You mean to use the hello_pio program, only need to specify HELLO_PIO_LED_PIN pin to work properly? But after I modified it, I still can't make the pin greater than 32 work properly, and this is the case for both RP2350B. I updated the latest SDK and sample program.

peterharperuk, 2024-11-20:
Yes, this program seems to work for me if I use pin 45. What board are you using? Can you double check that HELLO_PIO_LED_PIN is set correctly and PICO_PIO_USE_GPIO_BASE is true? I will add the following to the test.

#if HELLO_PIO_LED_PIN >= 32 && !PICO_PIO_USE_GPIO_BASE
#error Attempting to use a pin>32 on a platform that doesn't support it
#endif

Octopus1633, 2024-11-21:
The PICO_PIO_USE_GPIO_BASE output was always 0, but when I specified 1, everything worked fine!

kilograham, 2024-11-21:
If `PICO_PIO_USE_GPIO_BASE==0`, then `PICO_RP2350A` is probably set in your board header which is bad if you have a RP2350B... I would fix the latter, since it may affect other things that use GPIO > 32 (we omit extra code for the higher pins when we know they aren't there)

peterharperuk, 2024-11-21:
It's not very obvious I guess. I'll add some checks to the examples which will hopefully clarify things.