raspberrypi/pico-sdk #2059: Add DMA APIs for count modes

kilograham, 2024-11-19:
There is no ability to configure the Count Modes field of the DMA`s CHx_TRANS_COUNT register for RP2350 - its processing is missing in the SDK API version 2.0.0.

From RP2350 datasheet:
[image]
[image]

File "~\\.pico-sdk\sdk\2.0.0\src\rp2_common\hardware_dma\include\hardware\dma.h"
Actual code:
```
static inline void dma_channel_set_trans_count(uint channel, uint32_t trans_count, bool trigger) {
    if (!trigger) {
        dma_channel_hw_addr(channel)->transfer_count = trans_count;
    } else {
        dma_channel_hw_addr(channel)->al1_transfer_count_trig = trans_count;
    }
}
```
Expected code example:
```
static inline void dma_channel_set_trans_count(uint channel, uint32_t mode, uint32_t trans_count, bool trigger) {
    assert(mode == DMA_CH0_TRANS_COUNT_MODE_VALUE_NORMAL || mode == DMA_CH0_TRANS_COUNT_MODE_VALUE_TRIGGER_SELF || mode == DMA_CH0_TRANS_COUNT_MODE_VALUE_ENDLESS);
    assert(trans_count < (1 << DMA_CH0_TRANS_COUNT_MODE_LSB));
    if (!trigger) {
        dma_channel_hw_addr(channel)->transfer_count = trans_count;
    } else {
        dma_channel_hw_addr(channel)->al1_transfer_count_trig = trans_count;
    }
    dma_channel_hw_addr(channel)->transfer_count |= mode << DMA_CH0_TRANS_COUNT_MODE_LSB;
}
```

TheLostLambda, 2024-12-28:
Also running into this issue — any workarounds for the time being? Or is this working on the development branch yet?

Or alternatively, is this something that needs help?

TheLostLambda, 2024-12-28:
For what it's worth, this didn't work for me outside of the SDK:
```c
 dma_channel_hw_addr(channel)->transfer_count |= mode << DMA_CH0_TRANS_COUNT_MODE_LSB;
```
I found that reading `dma_channel_hw_addr(channel)->transfer_count` (or `dma_hw->ch[dma_chan].transfer_count` in my code) always returned zero until *after* the channel was started / triggered, at which point updating the transfer count had no effect. I needed to set it before `dma_channel_start()`, but I couldn't read it until after `dma_channel_start()`.

To avoid this issue, I ended up with something that just directly wrote the final value to `transfer_count`:
```c
dma_channel_hw_addr(channel)->transfer_count = trans_count | mode << DMA_CH0_TRANS_COUNT_MODE_LSB;
```

TheLostLambda, 2024-12-28:
Happy to write a PR for this, but it would be good to know if people think we should update `dma_channel_configure` too, since that is another common route for setting `transfer_count`?

kilograham, 2025-07-22:
> For what it's worth, this didn't work for me outside of the SDK:
> 
>  dma_channel_hw_addr(channel)->transfer_count |= mode << DMA_CH0_TRANS_COUNT_MODE_LSB;
> 
> I found that reading `dma_channel_hw_addr(channel)->transfer_count` (or `dma_hw->ch[dma_chan].transfer_count` in my code) always returned zero until _after_ the channel was started / triggered, at which point updating the transfer count had no effect. I needed to set it before `dma_channel_start()`, but I couldn't read it until after `dma_channel_start()`.
> 
> To avoid this issue, I ended up with something that just directly wrote the final value to `transfer_count`:
> 
> dma_channel_hw_addr(channel)->transfer_count = trans_count | mode << DMA_CH0_TRANS_COUNT_MODE_LSB;

Do the existing APIs work correctly if you pre-encode the trans_count?

I'm thinking i may just document the trans_count as being pre-encoded, and add helpers to produce a
* unlimited (RP2350)
* self-triggering (RP2350)
* normal 
argument