raspberrypi/pico-sdk #2081: Can't configure where local picotool binary is

cinderblock, 2024-11-21:
I have the `picotool` binary on my path. `cmake` (and `Findpicotool.cmake`) still somehow can't seem to find `picotool`.

They try to build it, but that fails for other reasons. I don't need to build it. I have a working binary. Why can't it just use it?

I have also tried following [picotool's instructions](https://github.com/raspberrypi/picotool?tab=readme-ov-file#usage-by-the-raspberry-pi-pico-sdk) on telling the SDK where the single shared picotool is by setting the `picotool_DIR` environment variable to the folder that contains the binary (or with `-D...` to cmake`) and it still tries to build.

I must be doing something silly...

will-v-pi, 2024-11-21:
Have you followed the instructions to **install** picotool in your custom location? The SDK is not actually looking for the picotool binary, it’s looking for the `picotoolConfig.cmake` file which only gets created when you **install** picotool.

The `develop` branch of picotool also has a more fleshed out README on this, so might be worth looking at instead of `master` if you’re having issues.

Also, as a side note, `Findpicotool.cmake` isn’t actually used to find picotool, it’s used to build it if it can’t be found - the actual finding is done by the CMake `find_package` function.

cinderblock, 2024-11-21:
I downloaded the binary for my system from github and put it on my path. Yes, that is not the default install instruction, but I didn't want to build picotool for my system because that's broken for other reasons and I just wanted to use a binary. Figured that should be easy.

In the interim, it looks like the default install instructions might automatically download a binary instead of building (if available) so I should give that another go I guess. Still, I feel like being more clear about how `picotool`'s location can be configured should be a little more obvious, no? Are we ignoring the path too?

will-v-pi, 2024-11-21:
Yes, the binary in the path is not used - CMake’s `find_package` is just looking for the `picotoolConfig.cmake` file which then points at the `picotoolTarget.cmake` file, which points at the picotool binary. Hopefully whatever you downloaded came with those files, so if you just use that downloaded directory as-is and set `picotool_DIR` to that directory it should work - no need to touch the `PATH`.

cinderblock, 2024-11-24:
I see those files now. I downloaded this from raspberry pi's repo: https://github.com/raspberrypi/pico-sdk-tools/releases

I'm not seeing how then to put picotool on the path and supporting what cmake expects. I guess I could symlink to the binary, but that gets weird on non-linux systems.

I still think the original requests here stands - or at least to get error messages that more helpfully point to the solution.

Is there some reason we couldn't just specify the binary directly?

will-v-pi, 2024-11-24:
If you just add that directory you downloaded to your PATH variable (eg in `.bashrc` on Linux) then that’s the easiest way to add picotool to your path - you could also symlink it as you suggest.

u77345, 2024-12-17:
What made it work on MacOS:
When followed the instructions on https://github.com/raspberrypi/picotool/blob/develop/README.md#usage-by-the-raspberry-pi-pico-sdk to install to the standard location with `make install`, the `~/.local/bin` had to be added to the `PATH`. Also pointed out in the instructions, just not prominent enough.

will-v-pi, 2025-02-03:
Closing this as not planned, as we don't plan to support specifying the picotool binary directly - the existing CMake method works fine (provided you read and follow the instructions, specifically the installation step), and ensures version compatibility.

If there are issues in the picotool documentation for this, or it needs more clarification, then feel free to raise a new issue or PR in the picotool repository with suggestions.

closed 2025-02-03 by will-v-pi

sdbbs, 2025-02-20:
> Yes, the binary in the path is not used - CMake’s `find_package` is just looking for the `picotoolConfig.cmake` file which then points at the `picotoolTarget.cmake` file, which points at the picotool binary.

Are you sure it is called `picotoolTarget.cmake`, and not <code>picotoolTarget<b>s</b>.cmake</code>? Searching in vain for `picotoolTarget.cmake` while being confused where it might be located ended up sucking some hours from my life ...

> If there are issues in the picotool documentation for this, or it needs more clarification, then feel free to raise a new issue or PR in the picotool repository with suggestions.

I would say this is a documentation problem - so I've opened https://github.com/raspberrypi/pico-sdk/issues/2308