raspberrypi/pico-sdk #2091: No support for Pimoroni Pico Plus 2 W

gonzalolarralde, 2024-11-24:
[This board](https://shop.pimoroni.com/products/pimoroni-pico-plus-2-w?variant=42182811942995) combines an RP2350B with the same wireless chip in Pico W.

There is not yet a board definition for it. Based on the [schematics](https://cdn.shopify.com/s/files/1/0174/1800/files/Pimoroni_Pico_Plus_2_W_Schematic.pdf?v=1727350279) we can know that they wire an RM2 (Infineon CYW43439) using the same pins used on Pico W and the LED is routed through the chip as well.

Slion, 2024-11-24:
I was told to use the dev branch and compile using `set(PICO_BOARD "pico_w")`.

lurch, 2024-11-25:
Fixed in `develop` (and also `master`! :tada: )

lurch, 2024-11-25:
> I was told to use the dev branch and compile using:
> 
> ```
> set(PICO_PLATFORM "rp2350")
> set(PICO_BOARD "pico_w")
> ```
> 
> That does not work with SDK 2.0.0 main branch.

That definitely won't work, because `pico_w` uses the `rp2040`, not the `rp2350`.

PR #2092 (2024-11-25): Add board file for Pimoroni Pico Plus 2 W
See the board details here: [Pimoroni Pico Plus 2 W](https://shop.pimoroni.com/products/pimoroni-pico-plus-2-w?variant=42182811942995)

This board is similar to pimoroni_pico_plus2_rp2350. I cannot reuse the same header file as this one does not have a direct wire to the LED, similar to pico_w it depends on the WiFi chip for it (compatibility, presumably).

I have no affiliation with Pimoroni, I just purchased a board and I'm using it with this modifications. I see that @ZodiusInfuser has posted some PRs for Pimoroni boards in the past so maybe they ca [... cut]