raspberrypi/pico-sdk #2097: Bazel CI checks are broken

lurch, 2024-11-25:
```
Run bazel build @picotool//:picotool
  bazel build @picotool//:picotool
  shell: /usr/bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.10.15/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.10.15/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.10.15/x64
    Python[2](https://github.com/raspberrypi/pico-sdk/actions/runs/12007459038/job/33468066168#step:7:2)_ROOT_DIR: /opt/hostedtoolcache/Python/[3](https://github.com/raspberrypi/pico-sdk/actions/runs/12007459038/job/33468066168#step:7:3).10.15/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.10.15/x6[4](https://github.com/raspberrypi/pico-sdk/actions/runs/12007459038/job/33468066168#step:7:4)
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.10.1[5](https://github.com/raspberrypi/pico-sdk/actions/runs/12007459038/job/33468066168#step:7:5)/x64/lib
INFO: Invocation ID: 19058626-0ad9-4073-81f3-15bd245e04c[6](https://github.com/raspberrypi/pico-sdk/actions/runs/12007459038/job/33468066168#step:7:6)
Computing main repo mapping: 
ERROR: Error computing the main repository mapping: module not found in registries: picotool@2.1.0
Error: Process completed with exit code 4[8](https://github.com/raspberrypi/pico-sdk/actions/runs/12007459038/job/33468066168#step:7:8).
```

ping @armandomontanez

lurch, 2024-11-25:
Possibly related to https://github.com/raspberrypi/picotool/pull/181 ?

armandomontanez, 2024-11-25:
Yeah, I need to add the new release to the [BCR](https://github.com/bazelbuild/bazel-central-registry/) so the reference can resolve. I'll get that sorted out ASAP.

armandomontanez, 2024-11-25:
PR out at https://github.com/bazelbuild/bazel-central-registry/pull/3253

lurch, 2024-11-25:
Please reply here when that has been merged, and I'll close this issue :slightly_smiling_face:  :chains:

armandomontanez, 2024-11-25:
PR has merged, Bazel build should be happy again.

lurch, 2024-11-26:
Fantastic, thank you!

closed 2024-11-26 by lurch

lurch, 2024-11-26:
> PR has merged, Bazel build should be happy again.

Confirmed - all the Bazel CI tests on #2098 are green :green_circle:

will-v-pi, 2024-12-02:
Is it also worth re-running the Bazel jobs for the SDK 2.1.0 release (https://github.com/raspberrypi/pico-sdk/actions/runs/12005305108/job/33461692064), so the master branch doesn't show a CI failure?

lurch, 2024-12-02:
> Is it also worth re-running the Bazel jobs for the SDK 2.1.0 release (https://github.com/raspberrypi/pico-sdk/actions/runs/12005305108/job/33461692064), so the master branch doesn't show a CI failure?

Good idea. I just clicked the button to do that.... :crossed_fingers: