raspberrypi/pico-sdk #2101: Pico 2 W - FreeRTOS - cyw43_arch_init within Task causes Panic

jondurrant, 2024-11-26:
I'm using Raspberry Pi version of FreeRTOS, current version which was last updated about 3 months ago. Linking: pico_cyw43_arch_lwip_sys_freertos and calling cyw43_arch_init from within a task. I get error as below. Any suggestions?

Error:
assertion "!(raw_irq_mask[get_core_num()] & gpio_mask)" failed: file "/Users/jondurrant/pico/pico-sdk-2.1
.0/src/rp2_common/hardware_gpio/gpio.c", line 222, function: gpio_add_raw_irq_handler_with_order_priority
_masked

This is a migration of code that runs flawlessly on Pico W.

jondurrant, 2024-11-26:
SDK is 2.1.0

peterharperuk, 2024-11-26:
Could you be calling cyw43_arch_init twice? That's the only way I could think that this could happen off the top of my head. I assume it's the cyw43_irq_init call that's causing the problem.

> This is a migration of code that runs flawlessly on Pico W.

I can't think why this would be different on Pico W.

jondurrant, 2024-11-26:
Certain this is only calling once at the top of the main thread.

kilograham, 2024-11-26:
a stack trace would be helpful

jondurrant, 2024-11-27:
Stack trace:

Thread #1 [rp2350.dap.core0] 1 (Name: rp2350.dap.core0, state: breakpoint) (Suspended : Signal : SIGTRAP:Trace/breakpoint trap)	
	_exit() at newlib_interface.c:45 0x100046dc	
	__assert_func() at newlib_interface.c:168 0x100047a6	
	gpio_add_raw_irq_handler_with_order_priority_masked() at gpio.c:222 0x10000a82	
	gpio_add_raw_irq_handler_with_order_priority() at gpio.h:674 0x1001789a	
	cyw43_irq_init() at cyw43_driver.c:56 0x1001789a	
	handle_sync_func_call() at async_context_freertos.c:189 0x10017dea	
	async_context_base_execute_once() at async_context_base.c:101 0x10008a36	
	process_under_lock() at async_context_freertos.c:49 0x10017cda	
	async_context_task() at async_context_freertos.c:72 0x10017dbc	
	0x100050c0	
Thread #2 [rp2350.dap.core1] 2 (Name: rp2350.dap.core1, state: debug-request) (Suspended : Container)	
	0xda	
	0x146

jondurrant, 2024-11-27:
So some further debugging on this I can confirm that cyw43_irq_init is being called twice. This is happening somewhere in the cyw43_arch_init code. Trying to get my head around why....

kilograham, 2024-11-27:
This might be related to SMP - does the problem go away if you turn off SMP?

jondurrant, 2024-11-27:
This is running single core.  To confirm I got the config correct the stack trace above shows no functions running on the second core.

jondurrant, 2024-11-27:
The multiple calls to cyw43_irq_init are from the async_context_task looping. It looks to me like the self->task_should_exit should have been set.

peterharperuk, 2024-11-27:
It seems odd that I don't see this with the pico-examples. Are you doing anything different to the examples? https://github.com/raspberrypi/pico-examples/blob/master/pico_w/wifi/freertos/iperf/picow_freertos_iperf.c#L103
I think that you will get SMP now if configNUMBER_OF_CORES > 1. Any chance you could post your code from main up to this point and your Freertos config header file?

jondurrant, 2024-11-27:
Of course let me post the code and config. It is very odd and I am only calling cyw43_arch_init in my trimmed down test. I am really scratching my head with what is going on. So very grateful for the help.
[bug.tar.zip](https://github.com/user-attachments/files/17933181/bug.tar.zip)

peterharperuk, 2024-11-27:
I think this is a bug. I'm a bit stumped why it only occurs when configNUMBER_OF_CORES = 1
In the following function...

```
uint32_t async_context_freertos_execute_sync(async_context_t *self_base, uint32_t (*func)(void *param), void *param) {
    async_context_freertos_t *self = (async_context_freertos_t*)self_base;
    hard_assert(xSemaphoreGetMutexHolder(self->lock_mutex) != xTaskGetCurrentTaskHandle());
    sync_func_call_t call;
    call.worker.do_work = handle_sync_func_call;
    call.func = func;
    call.param = param;
    call.sem = xSemaphoreCreateBinary();
    async_context_add_when_pending_worker(self_base, &call.worker);  <- WORKER IS PENDING ALREADY
    async_context_set_work_pending(self_base, &call.worker); <- WORKED MADE PENDING AGAIN.
```

The following might be the fix.

```
sync_func_call_t call = {0};
```

jondurrant, 2024-11-27:
Thanks Peter. Nice to know I am not going nuts.  I'll try that fix and also perhaps running under SMP. Kind of weird though, I can't work out how the work_pending is being reset.

jondurrant, 2024-11-27:
I can confirm that the following line fixes the issue
sync_func_call_t call = {0};

Thank you for diagnosing so quickly. Look forward to the official update.

rvt, 2024-12-21:
I can also confirm that this fixed this exact issue. Same update to 2.1.0

tlberglund, 2024-12-31:
Additional confirmation here. I've been chasing this and going crazy for about a week, found the same double-call as @jondurrant, and am now enjoying working code. Thank you for the fix! 💥

kilograham, 2025-01-11:
merged into develop

PR #2106 (2025-01-11): Initialise stack in async_context_execute_sync call
We need to do this to avoid work_pending being set twice.

Fixes #2101
commit 5202c272 (2025-01-11): Initialise stack in async_context_execute_sync call (#2106)

We need to do this to avoid work_pending being set twice.

Fixes #2101
closed 2025-01-11 by kilograham