raspberrypi/pico-sdk #2102: Marking binary with PICO_COPY_TO_RAM causes build process to remove data placed in .rodata/.flashdata section

XLuma, 2024-11-26:
For extra optimisation by reducing flash operations to a maximum, we added the `set(PICO_COPY_TO_RAM 1)` to our project's CMakeList.
The program's code includes read-only data that we include using .incbin, and placed in .rodata using `.section .flashdata` at the top of the assembly file.

According to linker scripts in pico-sdk, .flashdata sections are placed in .rodata, which is what happens. The output .elf has all the data in the correct place. However, the resulting .uf2 image is *way* smaller than it should be prior to adding PICO_COPY_TO_RAM. On one hand this makes sense because you'd only want to keep the actual program code that can fit in ram. But on the other hand, the documentation outlines how .flashdata will force variables and data to be kept in the final program and stored in flash.

Is there something we may be doing wrong with our usage of PICO_COPY_TO_RAM ? Is stripping .rodata intended and we should program the flash with read-only data another way ?

kilograham, 2025-02-06:
Are you sure your section is marked allocatable?

will-v-pi, 2025-02-06:
Could you provide some code that reproduces your issue? I can't reproduce it myself using SDK 2.1.0 - if I add a .S file to my compilation with the following contents:
```asm
.section .flashdata
.global pico_sdk_import
pico_sdk_import:
.incbin "pico_sdk_import.cmake"
.byte 0
```
then add a quick print to my main function
```c
extern char pico_sdk_import[];
printf(pico_sdk_import);
```
the included file gets included in the UF2 and the ELF file, and printed out at runtime, with a copy_to_ram binary.

XZCE, 2025-02-20:
I'm seeing the same thing with @will-v-pi 's example.  With ".section .flashdata" the data just disappears (from .elf and .bin files) and I get a NULL pointer in the disassembly like it was weak.  If I make it ".section .flashdata.test" then the .elf file does have the data, I get a pointer to it (a flash address) in the disassembly, but the .bin file still doesn't have the payload.

This is with the Windows toolchain.  Maybe it's only working properly in Linux?

XZCE, 2025-02-20:
Well, I've worked around it by using ".section .reset" so it ends up going in .flashtext instead of .rodata and that works for me...  But I can see .flashdata is meant to go into .rodata and .rodata is meant to go into .data...  And when I look at the "Sections" output at the top of the .dis file, I can see weird fragments of the same sections in different file locations...  So, all a bit non-standard perhaps, maybe the tools shouldn't assume all the data is "together" in the file?  Dunno.

For the record, moving my 20KB of read-only data into .reset made almost no difference to startup time.  But this is after I changed the rosc clock to div1 before the copy-to-ram loops, which makes thing much faster.  (And is why I'm here.)