raspberrypi/pico-sdk #2103: Order of operations in `mbedtls_pkcs5_pbkdf2_hmac` does not release pico's SHA256 hardware lock

nilswiersma, 2024-11-27:
The current way that the SHA256 hardware engine and the mbedtls implementation of `mbedtls_pkcs5_pbkdf2_hmac` interact with each other results in the situation where that function never releases the SHA256 engine lock. Below I included a trace of how functions are called in [pico_mbedtls.c](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_mbedtls/pico_mbedtls.c) and in mbedtls' [pkcs5.c](https://github.com/Mbed-TLS/mbedtls/blob/2ca6c285a0dd3f33982dd57299012dacab1ff206/library/pkcs5.c#L268).

You can see that it ends with a call to `mbedtls_md_hmac_reset` which claims the lock, but then it is never released. I would submit a pull request to address this, but I am not sure if it this a problem of mbedtls, or the link to the pico's SHA256 engine? 

```
Iterations: '3'
Password: 'password' (length 8)
Salt: 686d6163dea2f2bf251b2275
KDF START

mbedtls_pkcs5_pbkdf2_hmac
mbedtls_md_hmac_starts
mbedtls_sha256_starts_ret
pico_sha256_lock
mbedtls_sha256_update_ret
mbedtls_md_hmac_update1
mbedtls_sha256_update_ret
mbedtls_md_hmac_update2
mbedtls_sha256_update_ret
mbedtls_md_hmac_finish1
mbedtls_sha256_finish_ret
pico_sha256_unlock
mbedtls_sha256_starts_ret
pico_sha256_lock
mbedtls_sha256_update_ret
mbedtls_sha256_update_ret
mbedtls_sha256_finish_ret
pico_sha256_unlock
mbedtls_md_hmac_reset1
mbedtls_sha256_starts_ret
pico_sha256_lock
mbedtls_sha256_update_ret
mbedtls_md_hmac_updater3
mbedtls_sha256_update_ret
mbedtls_md_hmac_finish2
mbedtls_sha256_finish_ret
pico_sha256_unlock
mbedtls_sha256_starts_ret
pico_sha256_lock
mbedtls_sha256_update_ret
mbedtls_sha256_update_ret
mbedtls_sha256_finish_ret
pico_sha256_unlock
mbedtls_md_hmac_reset2
mbedtls_sha256_starts_ret
pico_sha256_lock
mbedtls_sha256_update_ret
mbedtls_md_hmac_updater3
mbedtls_sha256_update_ret
mbedtls_md_hmac_finish2
mbedtls_sha256_finish_ret
pico_sha256_unlock
mbedtls_sha256_starts_ret
pico_sha256_lock
mbedtls_sha256_update_ret
mbedtls_sha256_update_ret
mbedtls_sha256_finish_ret
pico_sha256_unlock
mbedtls_md_hmac_reset2
mbedtls_sha256_starts_ret
pico_sha256_lock
mbedtls_sha256_update_ret

KDF took 111ms
```

peterharperuk, 2024-11-27:
It does look a bit like mbedtls_sha256_free should release the lock if needed. Is that your fix? PR would be welcome.

nilswiersma, 2024-11-27:
At the moment my fix is something hacky in the application that uses this library. Your suggestion sounds great, thanks! I will work on a PR.

peterharperuk, 2025-02-03:
I believe this is fixed

PR #2105 (2025-01-31): Release the lock in mbedtls_sha256_free if it was not released already
Release the lock in `mbedtls_sha256_free` if it was not released already. Requires making `pico_sha256_unlock` include-able. Fixes #2103.
commit 6c06f90b (2025-01-31): Release the lock in mbedtls_sha256_free if it was not released already (#2105)

* Release the lock in mbedtls_sha256_free if it was not released already. Requires making pico_sha256_unlock include-able. Fixes #2103.

* addres lock release API comment

* Fix description of pico_sha256_cleanup

---------
closed 2025-02-03 by peterharperuk