raspberrypi/pico-sdk #2112: "// pico_cmake_set" not seen in included header files.

Memotech-Bill, 2024-11-28:
If a board header file includes other header files, then CMake does not see any **pico_cmake_set** directives in the secondary header file.

This particularly affects the **vgaboard.h** header file, which includes **pico.h**. None of the pico settings are seen by CMake. This does not matter in this case as all the settings are default values anyway.

However, it is not possible to simply change **pico.h** to **pico2.h** for a VGA board with pico2 installed, as CMake will still think it is compiling for RP2040 rather than RP2350.

lurch, 2024-11-28:
FWIW, https://github.com/raspberrypi/pico-sdk/blob/develop/tools/check_board_header.py _does_ look at `#inclue` files :laughing:

lurch, 2024-11-28:
I think the search for `#include` lines would need to be recursive though, since you might well have nested `#include`s?

will-v-pi, 2024-11-28:
That's true - in which case maybe add the parsing into the `while(HEADER_FILE_CONTENTS)` loop, and append to `HEADER_FILE_CONTENTS` when you see an include directive. This would also have the effect of processing the `cmake_set_default`s from the included header after the new header, which I think is the preferred way round (so you can override them in the new header)?

lurch, 2024-11-28:
I guess _appending_ to `HEADER_FILE_CONTENTS` (rather than doing an "in-place insert") might theoretically cause a problem if you had something like:

`file1.h`
```
// pico_cmake_set VAR1=VAL1
// pico_cmake_set VAR2=VAL2
#include "file2.h"
// pico_cmake_set VAR3=VAL3
```

`file2.h`
```
// pico_cmake_set VAR2=VAL4
// pico_cmake_set VAR3=VAL5
// pico_cmake_set VAR4=VAL6
```

I think you'd "expect" `VAR1=VAL1, VAR2=VAL4, VAR3=VAL3, VAR4=VAL6` but if you only appended to `HEADER_FILE_CONTENTS` when encountering an `#include` then you'd actually end up with `VAR1=VAL1, VAR2=VAL4, VAR3=VAL5, VAR4=VAL6` ?
But this may not _actually_ be a problem, and doing a simple append-only approach might be "good enough" for the cases where `pico_cmake_set` is actually used?

will-v-pi, 2024-11-28:
That's very true (with the opposite expected for `cmake_set_default`) - so it would be better to pre-pend the new file to `HEADER_FILE_CONTENTS` (the same as inserting in-place as we're popping from the front), which should be simple enough.

I'll steal your regex from `check_board_header.py` and whip up a PR

Memotech-Bill, 2024-11-28:
I have wondered whether BOARDS could optionally be a list, rather than a single filename.

In that way, a pico2_w on a VGA board might be defined as:

````
BOARD=pico2_w,vgaboard
````

That would save having to have multiple definitions for any of the boards which will accept any flavour of pico.

lurch, 2024-11-28:
> I have wondered whether BOARD could optionally be a list, rather than a single filename.

Ugh no. That might lead to people thinking that `BOARD=pico2,pico_w` is the same thing as `BOARD=pico2_w`.

I guess some concept of `CARRIER_BOARD` might be useful (see e.g. #696 and #837 ), but I guess that's a much bigger change.

lurch, 2024-11-28:
@Memotech-Bill Would you like to give #2113 a try please, and see if it behaves as you expect?

Memotech-Bill, 2024-11-28:
> we intend to make a PICO_CARRIER_BOARD (or CARRIED board) to cover those, which could have helped, but will think about recursion too... In this case probably a good idea

This should perhaps be PICO_ADDON_BOARDS, and should probably have the capability to be a list. Unlike the RPi, Pico addon boards do not have configuration EEPROMs, so it is straightforward to stack multiple addon boards.

I can't call them HATs as the usually go underneath the Pico, not on top.