raspberrypi/pico-sdk #2114: Assembling compile_time_choice.S does not respect PICO_BOARD_HEADER_DIRS

Memotech-Bill, 2024-11-28:
A board header file may include another (secondary) header file.

Providing the secondary header file is in **pico-sdk/src/boards/include**, then everything works.

However if the secondary header file is in a user defined folder, which is specified in **PICO_BOARD_HEADER_DIRS**, then the compilation of **compile_time_choice.S** fails:

````
[  1%] Building ASM object pico-sdk/src/rp2350/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.o
cd /home/pi/pico/PicoBB/console/pico_w/build/pico-sdk/src/rp2350/boot_stage2 && /usr/bin/arm-none-eabi-gcc -DLIB_BOOT_STAGE2_HEADERS=1 -DPICO_32BIT=1 -DPICO_BOARD=\"vgaboard_cut_2w\" -DPICO_BUILD=1 -DPICO_FLASH_SIZE_BYTES="(2 * 1024 * 1024)" -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_RP2350=1 -I/home/pi/pico/pico-sdk/src/rp2350/boot_stage2/asminclude -isystem /home/pi/pico/pico-sdk/src/rp2350/hardware_regs/include -isystem /home/pi/pico/pico-sdk/src/rp2_common/hardware_base/include -isystem /home/pi/pico/pico-sdk/src/common/pico_base_headers/include -isystem /home/pi/pico/PicoBB/console/pico_w/build/generated/pico_base -isystem /home/pi/pico/pico-sdk/src/boards/include -isystem /home/pi/pico/pico-sdk/src/rp2350/pico_platform/include -isystem /home/pi/pico/pico-sdk/src/rp2_common/pico_platform_compiler/include -isystem /home/pi/pico/pico-sdk/src/rp2_common/pico_platform_panic/include -isystem /home/pi/pico/pico-sdk/src/rp2_common/pico_platform_sections/include -isystem /home/pi/pico/pico-sdk/src/rp2_common/hardware_dcp/include -isystem /home/pi/pico/pico-sdk/src/rp2350/hardware_structs/include -isystem /home/pi/pico/pico-sdk/src/rp2_common/hardware_rcp/include -isystem /home/pi/pico/pico-sdk/src/rp2350/boot_stage2/include -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -g -O3 -DNDEBUG -o CMakeFiles/bs2_default.dir/compile_time_choice.S.o   -c /home/pi/pico/pico-sdk/src/rp2350/boot_stage2/compile_time_choice.S
In file included from /home/pi/pico/PicoBB/console/pico_w/build/generated/pico_base/pico/config_autogen.h:7,
                 from /home/pi/pico/pico-sdk/src/common/pico_base_headers/include/pico/config.h:19,
                 from /home/pi/pico/pico-sdk/src/rp2350/boot_stage2/include/boot_stage2/config.h:12,
                 from /home/pi/pico/pico-sdk/src/rp2350/boot_stage2/compile_time_choice.S:11:
/home/pi/pico/PicoBB/boards/vgaboard_cut_2w.h:91:10: fatal error: boards/pico2+w.h: No such file or directory
 #include "boards/pico2+w.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
````

My user defined directory:
````
pi@raspberrypi:~/pico/PicoBB/console/pico_w $ ls ../../boards/
pico2+w.h  vgaboard_cut_2w.h   vgaboard_cut.h    vgaboard_sd.h    vgaboard_serial.h
pico+w.h   vgaboard_cut_2w.h~  vgaboard_cut_w.h  vgaboard_sd_w.h  vgaboard_serial_w.h
````

**config_autogen.h**
````
// AUTOGENERATED FROM PICO_CONFIG_HEADER_FILES and then PICO_<PLATFORM>_CONFIG_HEADER_FILES
// DO NOT EDIT!

// based on PICO_CONFIG_HEADER_FILES:

#include "/home/pi/pico/PicoBB/boards/vgaboard_cut_2w.h"
#include "/home/pi/pico/pico-sdk/src/rp2_common/cmsis/include/cmsis/rename_exceptions.h"

// based on PICO_RP2350_ARM_S_CONFIG_HEADER_FILES:
````

lurch, 2024-11-29:
@will-v-pi Would a similar thing also affect #2113 ?

Memotech-Bill, 2024-11-29:
> @will-v-pi Would a similar thing also affect #2113 ?

I don't think so. If it did I would have obtained a failure in CMake before getting to compiling compile_time_choice.S.

lurch, 2024-11-29:
I assumed that you **weren't** testing #2113 with custom header files in a user-defined `PICO_BOARD_HEADER_DIRS` location? :shrug: 

Perhaps I should have made the question in my previous comment more explicit: if there's an `#include` of a custom header file  in a user-defined `PICO_BOARD_HEADER_DIRS` location, and that custom header file contains `// pico_cmake_set` declaations, will those declarations get included or ignored by the code in #2113 ?

Memotech-Bill, 2024-11-29:
> I assumed that you **weren't** testing #2113 with custom header files in a user-defined `PICO_BOARD_HEADER_DIRS` location? 🤷

For my first test, the primary board header file was in a user defined location, but the secondary board header file was **pico2_w.h** in the standard location. That all worked, hence my [report of success](https://github.com/raspberrypi/pico-sdk/issues/2112#issuecomment-2506341498).
> Perhaps I should have made the question in my previous comment more explicit: if there's an `#include` of a custom header file in a user-defined `PICO_BOARD_HEADER_DIRS` location, and that custom [...]

I then went on to do a test with both primary and secondary board header files in a user defined location. I could see from the CMake output that this had correctly picked up the **pico_cmake_set** declarations from the secondary header file but then failed on compiling **compile_time_choice**.

will-v-pi, 2024-11-29:
> if there's an `#include` of a custom header file in a user-defined `PICO_BOARD_HEADER_DIRS` location, and that custom header file contains `// pico_cmake_set` declaations, will those declarations get included or ignored by the code in #2113 ?

They will still be included by #2113 - it searches all of `PICO_BOARD_HEADER_DIRS` in the same way that the SDK currently does when searching for included board files.

kilograham, 2025-02-03:
I think you should just use `#include "pico2+w.h"`

Memotech-Bill, 2025-02-03:
Interesting.

I had used the `boards/` prefix because the stock **vgaboard.h** file has:

````
#include "boards/pico.h"
````

A quick test suggests that omitting the `boards/` prefix works whether the nested include file is in either the SDK default folder or the user defined one.

My test is not definitive because I have re-arranged my code to avoid the need of having nested includes, and quickly hacking one back in results in a compilation failure for other causes. But I don't get any "No such file or directory" errors for the nested include file.

Memotech-Bill, 2025-02-03:
> A quick test suggests that omitting the `boards/` prefix works whether the nested include file is in either the SDK default folder or the user defined one.

My test was too quick. If the nested include file is in the SDK default directory, then you do have to prefix the file name by the `boards/` path, whereas if in the user defined the prefix should be omitted.

It would be nice if the behaviour was consistent between the two locations.

lurch, 2025-02-20:
> If the nested include file is in the SDK default directory, then you do have to prefix the file name by the `boards/` path, whereas if in the user defined folder, the prefix should be omitted.

IMHO that makes sense.

> It would be nice if the behaviour was consistent between the two locations.

Do you get the consistency you're after if you create a `boards` directory inside your user-defined folder?

Memotech-Bill, 2025-02-21:
The way to make the custom board files consistent with the SDK, without breaking any existing builds would be:

* **cmake** searches `$(PICO_BOARD_HEADER_DIRS)/include/boards` and then `$(PICO_BOARD_HEADER_DIRS)` for board header files.
* The C include folders includes `$(PICO_BOARD_HEADER_DIRS)/include` and `$(PICO_BOARD_HEADER_DIRS)`.

lurch, 2025-02-21:
If you don't mind me asking, why is this "consistency" so important to you? Perhaps the different include-styles of:
```c
#include "boards/pico.h"
#include "pico2+w.h"
```
makes it more obvious that `pico.h` is getting included from the SDK, but `pico2+w.h` is getting included from your user-defined directory? 🤷‍♂

Memotech-Bill, 2025-02-21:
I suppose just because I got bitten by it not being obvious that I needed to use the different style when including files from my user defined directory.

If it was not obvious to me, how many others will encounter the same issue?

But on further thought, my proposal of having `boards/include/boards`, while being consistent, is not obvious either.

The hack that would probably work for most people who make the same mistake as I did would be to have `$(PICO_BOARD_HEADER_DIRS)/..` as an include folder.