raspberrypi/pico-sdk #2118: Problem with repeating timer in SDK2.1.0

richard9999999999, 2024-12-01:
Hello,
I am experiencing a problem with repeating timer in SDK 2.1.0 (I haven't seen it in SDK 1.x.x before). The problem seems happen only when timer callback function is "light" (I mean doesn't contain too much of processing):

```
repeating_timer_t my_sampling_timer;

bool my_timer_callback(repeating_timer_t *rt)
{
	//volatile int tmp = 0;
	//for(int i=0; i <10000; i++)
	//{
	//	tmp++;
	//}
	
	return true;
}

int main()
{	
	alarm_pool_t *pool = alarm_pool_create(2, 1);	
	alarm_pool_add_repeating_timer_ms(pool, -20, my_timer_callback, NULL, &my_sampling_timer);

	while(1)
	{
		printf("looping\n");
	}
}
```
If the code is as above, the callback is called only few times. If I uncomment commented part in the callback (to make functionality to spend some more time in callback), the callback is called still during main looping (as desired). 
The problem happens in the RELEASE build only.
I use GCC 10.3.1 (not sure if related, probably not)

lurch, 2024-12-01:
Possibly similar to #737 ?

richard9999999999, 2024-12-01:
> Possibly similar to https://github.com/raspberrypi/pico-sdk/issues/737 ?
I hope no since that problem is almost 3 years old :-)
The problem seems to be easily reproducible form me using the simple code above. Please, could anybody else try that?

kilograham, 2024-12-01:
i did try it, and it worked fine. how are you telling if it is firing if empty?

kilograham, 2024-12-01:
note: it certainly might be timing dependent so exact code/compiler might be pertinent

richard9999999999, 2024-12-02:
> i did try it, and it worked fine. how are you telling if it is firing if empty?
I put breakpoint on line
`return true;`
in above code. It is minimalictics version to get the problem, originally I had a bit more complex timer callback code which looks cca as follows:
```
bool my_timer_callback(repeating_timer_t *rt)
{
	if(task_enabled)
	{
		runTask();
	}
	return true; 	//keep repeating
}
```
And I've found that as soon as I set task_enabled to false, task is not run even it it is enabled later. I spent some time trying to find the root of the problem (including studying generated disassembly code) and I found that really it seems that timer is stopped to be called after the time spent in the last timer callback is "too short". E.g following callback code:
```
bool my_timer_callback(repeating_timer_t *rt)
{
	if(task_enabled)
	{
		runTask();
	}
	else
	{
		sleep_ms(1);
	}
	return true; 	//keep repeating
}
```
works as expected.

richard9999999999, 2024-12-02:
As I wrote, I am getting the problem in RELEASE build only so you will need to set symbols (debug info) to be generated also in RELEASE build to be able to put the breakpoint there..

kilograham, 2024-12-02:
> As I wrote, I am getting the problem in RELEASE build only so you will need to set symbols (debug info) to be generated also in RELEASE build to be able to put the breakpoint there..

What you didn't write, but have now, is that you were checking with a breakpoint, which is exactly why I asked.

see also https://github.com/raspberrypi/pico-feedback/issues/428

kilograham, 2024-12-02:
I tried with a counter incremented in the timer callback, and printed out, in the main loop, and that worked fine

richard9999999999, 2024-12-02:
You are right, more complex example is needed to reproduce the problem (and not usage of breakpoint to detection of callback firing). I tweaked my problem more and found following:

peterharperuk, 2024-12-03:
Running fine for me on a Pico 2. By release build, do you mean you are just running cmake with no arguments?

peterharperuk, 2024-12-03:
I seem to be able to reproduce this if I load the program with openocd or drop the uf2 onto the pico.

richard9999999999, 2024-12-03:
By release build, do you mean you are just running cmake with no arguments

I mean cmake build type "Release" so compile with optimization for speed, however I am not if only this optimization mode is relevant here, potentially also the debug mode can have a specific preprocessor macro value defined (e.g. DEBUG. or _DEBUG or something like that, I didn't check) and preprocessor usually generates slightly different also c++ code)

peterharperuk, 2024-12-03:
Okay. I can reproduce it with your code. Investigating...

polpo, 2024-12-05:
I'm also having a possibly related issue with alarms with SDK 2.1.0 but not 2.0.0. A git bisect shows it's introduced by commit bd5523cc6405ecf055f87b6b92ec62219c4e36c1 - Improve best_effort_wfe_or_timeout (#1822). This is on RP2040, overclocked to 366MHz, using the ARM GNU Toolchain 13.3.Rel1. At first glance it appears as if an alarm is not firing. I can dive in more if necessary to see if this issue is actually related.

peterharperuk, 2024-12-05:
@polpo it would be helpful to know if the problem occurs for you in debug builds

lurch, 2024-12-05:
> This is on RP2040, overclocked to 366MHz

@polpo Is there an (approximate) minimum frequency above which this problem occurs, and below which it doesn't occur?

peterharperuk, 2024-12-05:
I might have got my head around this. Here's the relevant snippet from the timer irq.

```
        if (earliest_target != -1) { // cancelled alarm has target of -1
            ta_set_timeout(timer, timer_alarm_num, earliest_target);
        }
        // check we haven't now passed the target time; if not we don't want to loop again
    } while ((earliest_target - (int64_t)ta_time_us_64(timer)) <= 0);
```

We have a pool with one repeating timer on it. The irq has gone off because the alarm time was reached. We called the callback and it has told us to call it back again in 20000us. earliest_target is 20000us in the future. However ta_set_timeout WILL NOT set an alarm if it thinks there's an earlier alarm already set.

We have reached ta_set_timeout so quickly that the time to the current alarm is zero. As the time till the alarm is less than the time till the new target we want, a new alarm is not set, which leaves us with no alarms.

richard9999999999, 2024-12-05:
Thank you Peter, I am impressed this pico SDK community, rare to see such one today..

kilograham, 2024-12-06:
merged into develop

PR #2127 (2024-12-06): ta_set_timeout can fail to set an alarm
If alarm_pool_irq_handler takes <1us between handling an alarm and calling ta_set_timeout then no alarms will be set as it will appear as if an earlier alarm is already armedi before the target time.

Make sure ta_set_timeout always leaves with an alarm set by checking the armed status.

Fixes #2118
commit 969f5895 (2024-12-06): ta_set_timeout can fail to set an alarm (#2127)

* ta_set_timeout can fail to set an alarm

If alarm_pool_irq_handler takes <1us between handling an alarm and
calling ta_set_timeout then no alarms will be set as it will appear as
if an earlier alarm is already armedi before the target time.

Make sure ta_set_timeout always leaves with an alarm set by checking the
armed status.

Fixes #2118

* ta_disable_irq_handler should unarm its timer
closed 2024-12-06 by kilograham