raspberrypi/pico-sdk #2121: Faster double-precision library for RISC-V Hazard3

Panda381, 2024-12-03:
I created faster double-precision library for RISC-V Hazard3 core. Maybe it could be optionally used in the SDK. The functions have a bit lower precision than libc and do not support subnormals.
science functions: https://github.com/Panda381/PicoLibSDK/blob/main/_sdk/src/sdk_double_riscv_sci.S
templates of science functions in C: https://github.com/Panda381/PicoLibSDK/blob/main/_sdk/src/sdk_double_riscv_sci.c
other double functions: https://github.com/Panda381/PicoLibSDK/blob/main/_sdk/src/sdk_double_riscv.S

kilograham, 2024-12-03:
we could potentially add this as an option; don't know if we would use it as default if it is sigificantly off in precision

lurch, 2024-12-03:
See also #2003

Panda381, 2024-12-04:
In most cases the inaccuracy relates to a possible deviation in the 1 lowest bit of the mantissa. Larger deviations of several bits of the mantissa can occur at the boundaries of bands where the result is produced by subtracting from the base value - such as log(0.9999...) or cos_deg(89.9999..). Since precision is required from the double library, an optional use would be more appropriate.
Also, it's not 100% certain that some value won't result in curve inconsistency - I've run tests, but it would be good if someone else checked the results.
I've attached my measurements comparing speeds with the libc library (along with the float-precision functions I've provided at issue #2003 ): 
[image]