raspberrypi/pico-sdk #2122: PIO SM refactor

daveythacher, 2024-12-03:
DMA allows you to get the DMA channel number from the claim API. PIO however makes you suggest and test the PIO state machine. 

Would it not be better to make the PIO SM API use a pio_hw_t reference? Technically DMA could work this way also.

daveythacher, 2024-12-03:
pio_claim_unused_sm works.

closed 2024-12-03 by daveythacher

daveythacher, 2024-12-04:
@kilograham Why didn't you name it pio_sm_claim_unused? Do you have any plans to create GPIO claim API?

lurch, 2024-12-04:
> Do you have any plans to create GPIO claim API?

A `claim_unused_gpio` doesn't seem very useful, given that you need to actually have stuff wired up to the GPIO pins to actually do something? :electric_plug:

daveythacher, 2024-12-04:
Even though it is static some people do dynamic things with it. This can be chaotic. Ideally they belong to a manager which passes or releases the IO resource. This would prevent hazards via dependency/sanity checks. Overall this requires good style to matter, but does the SDK need to forbid this?

Ideally the modules are decoupled. We claim the modules with minimal impact to other modules. Shared resources are singletons while non-shared are factories. (PIO has a strange singleton in the instruction memory.) Most of the GPIO multiplexer and clock tree are singletons which create dependencies.

Would have been interesting, if the RISC-V used 128-bit SIMD in RP235x.

kilograham, 2024-12-04:
see #1490