raspberrypi/pico-sdk #2124: CMake recommends FetchContent_MakeAvailable over FetchContent_Populate

LandryNorris, 2024-12-04:
According to [the CMake docs](https://cmake.org/cmake/help/git-stage/module/FetchContent.html#command:fetchcontent_populate), FetchContent_Populate is deprecated as of 3.30. When syncing the examples, the following is printed by CMake 3.30:

```
  Calling FetchContent_Populate(pico_sdk) is deprecated, call
  FetchContent_MakeAvailable(pico_sdk) instead.  Policy CMP0169 can be set to
  OLD to allow FetchContent_Populate(pico_sdk) to be called directly for now,
  but the ability to call it with declared details will be removed completely
  in a future version.
```

FetchContent_MakeAvailable was added in 3.14, so to maintain compatibility with 3.13, a version check would be needed.

will-v-pi, 2025-01-07:
Neither of these can simply be changed to MakeAvailable, as they both need to populate the content without adding it to the build. I will look at changing the calls to use the other signature for FetchContent_Populate which hasn't been deprecated, so we can get the same behaviour. I believe the other signature has been around since FetchContent_Populate was added in 3.11, so no version check will be necessary.

kilograham, 2025-01-21:
merged into develop

PR #2169 (2025-01-20): Remove deprecated uses of FetchContent_Populate
This removes the deprecated uses of FetchContent_Populate (see #2124)

The SDK cannot just use MakeAvailable as the files shouldn't be added to the build, so instead this uses the full signature for FetchContent_Populate which has not been deprecated. The FetchContent_Declare call is kept in `pico_sdk_import.cmake` else it will re-clone the SDK for every new build directory when `PICO_SDK_FETCH_FROM_GIT_PATH` is set, rather than re-using the cloned SDK in `PICO_SDK_FETCH_FROM_GIT_PATH`.

Fixes #2124
commit c54475da (2025-01-20): Remove deprecated uses of FetchContent_Populate (#2169)

* Remove deprecated uses of FetchContent_Populate

Cannot use MakeAvailable as the files shouldn't be added to the build, so instead uses the full signature for FetchContent_Populate

Fixes #2124

* Fix GIT_SUBMODULES_RECURSE for older CMake versions
closed 2025-01-21 by kilograham