raspberrypi/pico-sdk #2126: Some `boot_stage2` of RP2350 still reference to `SSI` and its registers

matsobdev, 2024-12-05:
Those files are:
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2350/boot_stage2/boot2_at25sf128a.S
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2350/boot_stage2/boot2_is25lp080.S
https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2350/boot_stage2/boot2_w25x10cl.S

I guess, those devices are not commonly used and I think hardware RP2350's flash support might take care of them or at least `boot2_generic_03h` will be compatible, so I would call it a minor issue.

lurch, 2024-12-05:
I guess this wasn't noticed earlier, because RP2350 builds don't use a `boot_stage2` by default; you have to explicitly request it.

matsobdev, 2024-12-05:
I've just noticed, they have:
```
// todo port to QMI
```
so I think it is on somebody's mind already :D

michael-ring, 2024-12-07:
boot_stage2 was used as default for RP2040, what is used now for RP2350?

lurch, 2024-12-08:
> boot_stage2 was used as default for RP2040, what is used now for RP2350?

On page 649 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf in the release-notes for SDK 2.0.0 it says:

**boot_stage2**
* There are now separate implementations for RP2040 and RP2350.
* A `boot_stage2` is not needed on RP2350, but one can be included via the define `PICO_EMBED_XIP_SETUP=1`.

You'll probably want to have a look at Chapter 5 and Section 12.14 of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf for more information.