raspberrypi/pico-sdk #2128: i2c_write_blocking() hanging on SDK2.1.0 (while works fine on 2.0.0 or earlier)

tjko, 2024-12-06:
After switching from SDK 2.0.0 to 2.1.0, I noticed that [ss_oled](https://github.com/bitbank2/Pi_Pico_C_Projects/tree/master/ss_oled) OLED panel library stopped working (it hangs when initializing the OLED panel).
This issue was clearly introduced in SDK 2.1.0, as issue is not present in 2.0.0 or 1.5.0.

I traced the hang to i2c_write_blocking() call:

[ss_oled/BitBang_I2C.c(322):](https://github.com/bitbank2/Pi_Pico_C_Projects/blob/b1f964735b8ddeaf480ec15a54d7ca97f3576103/ss_oled/BitBang_I2C.c#L322)
```
 rc = i2c_write_blocking(i2c0, iAddr, &u8Register, 1, true); // true to keep master control of bus 
```

(at least on  Pico W  the i2c_write_blocking() call never returns)

Curiously, after I changed this line to following, that fixed the problem:

```
rc = i2c_write_timeout_us(pI2C->i2c, iAddr, &u8Register, 1, true, 100000); // true to keep master control of bus
```

(also its not timing out but succeeding normally, so blocking write should not hang...)

peterharperuk, 2024-12-06:
@tjko If you're keen, you could try reverting a couple of changes? It would be helpful to know which caused the problem. My bet would be on the first. I have a SSD1306 but it'll be a couple of days before I can test it...

commit 68778b1135783f374155df4d609737cd3036b027
commit 3bc866395cc755f344d341a7f0b13b99d641cb0f

tjko, 2024-12-07:
@peterharperuk , I reverted the change https://github.com/raspberrypi/pico-sdk/commit/68778b1135783f374155df4d609737cd3036b027 and fixed the hang as well.

I have two devices on the I2C bus:

0x38 AHT21  (Temperature/Pressure sensor)
0x3c SH1107 (1.5" 128x128 OLED Panel)

I also noticed that if I only have the OLED panel connected, I don't see the 'hang'...

peterharperuk, 2024-12-07:
Thanks. Do you know if the lockup occurs if you just have the AHT21 connected?

tjko, 2024-12-10:
No, since lockup happens in the ss_oled library, and when there is no panel found in the bus it won't get far enough to initialize it...

I noticed that the sensor library uses 'non-blocking' functions (i2c_write_timeout_us(), i2c_read_timeout_us()), so new SDK didn't cause any issues with it.

I'm initializing I2C sensors first, before trying to initialize the OLED panel, maybe the condition is triggered by first accessing another device with "non-blocking" read/write calls and then trying to initialize OLED panel, where the library is using the "blocking" read/write calls...