raspberrypi/pico-sdk #2132: Program stalled when started using openocd

alphatech56, 2024-12-09:
When I started program hello_serial provided as an example in sdk 2.1 on pico W using  **openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg -c "program hello_serial.elf verify reset exit"** I got “hello word” only once. After rebooting pico program runs correctly. 

With sdk 1.5 this commands runs correctly.

alphatech56, 2024-12-10:
Sorry, everything was fine with sdk 2.0.0

Gerhard-Vienna, 2024-12-22:
I can confirm that the problem doesn't occur under SDK 2.0.0, but a downgrade is not a solution, at best it's a workaround.
In 'Hardware timer not working after reset under debugger #1152', I found that adding 
`timer_hw->dbgpause` = 0; `
to the code fixed the problem in SDK 2.1.0.

This deletes bits 1 and 2 in the DBGPAUSE register. However, this seems to be a temporary solution as these registers are set by default after a reset (see RP2040 datasheet, Table 539 'DBGPAUSE Register'). 

The description of this register states: 'enable pause **when the corresponding debug ports are active'**. 
In my opinion, the problem therefore lies in the fact that with SDK 2.1.0 the debug ports are still active after the reset, which is probably a bug.

alphatech56, 2024-12-23:
Thank you for update

lurch, 2024-12-26:
From what I vaguely remember from previous GitHub issues about this, I think it _might_ be more to do with the OpenOCD version that you're using, than the SDK version you'e using? :man_shrugging:  Lots more info in https://github.com/raspberrypi/pico-feedback/issues/428

Gerhard-Vienna, 2024-12-27:
Thank you for your reply!

I know this thread. It led me to the DBGPAUSE register. However, it is mainly about what happens during debugging. The problem here occurs **after openocd has finished.**
I have both SDK 2.0.0 and 2.1.0 installed and the problem only occurs with SDK 2.1.0. I use the same openocd with both SDK versions:
```
openocd -v
Open On-Chip Debugger 0.12.0+dev-gebec9504d (2024-11-25-20:09)
Downloaded from https://github.com/raspberrypi/openocd.git
```

The parameters are also the same:
`openocd -s tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c 'adapter speed 5000' -c 'program blink.elf verify reset exit'
`
Does openocd need other parameters with SDK 2.1.0?
I have tried some ideas from raspberrypi/pico-feedback#428, but without success.

lurch, 2024-12-27:
Hopefully @peterharperuk will get a chance to look at this sometime in the new year.

Gerhard-Vienna, 2024-12-27:
Thanks for taking up this matter.

peterharperuk, 2025-01-02:
I can reproduce this.

peterharperuk, 2025-01-02:
Duplicate of https://github.com/raspberrypi/pico-sdk/issues/2118

closed 2025-01-02 by peterharperuk

Gerhard-Vienna, 2025-01-02:
I just tested it with the development branch. Problem solved, thanks!

peterharperuk, 2025-01-02:
Thanks for testing