raspberrypi/pico-sdk #2133: RP2350B pin PIO function abnormality

Octopus1633, 2024-12-09:
I have successfully run the hello_pio example on RP2350B based on #2051 . Now RP2350B has two pins connected to USB. I want to use PIO to simulate USB to realize USB master-slave communication, but after burning the firmware, I don't detect any signal on the pins. Even if I use pins below 32 and PICO_PIO_USE_GPIO_BASE=0.

https://github.com/sekigon-gonnoc/Pico-PIO-USB This code works fine on the RP2040. If it is run on the RP2350B and the pins below 32 are used to implement the PIO function, the program should run without any modification.

peterharperuk, 2024-12-09:
Does it work on a RP2350A?

lurch, 2024-12-09:
Perhaps you just need to modify the code to remove the GPIO isolation latches? (A new feature in RP2350 that wasn't in RP2040)

Octopus1633, 2024-12-10:
> Does it work on a RP2350A?

@peterharperuk I used RP2350A to test that USB communication can be carried out normally, and a stable signal can be detected.

Octopus1633, 2024-12-10:
> Does it work on a RP2350A?

@peterharperuk Hello, I used RP2350A to test that USB communication can be carried out normally, and a stable signal can be detected.

> Perhaps you just need to modify the code to remove the GPIO isolation latches? (A new feature in RP2350 that wasn't in RP2040)

@lurch Hello, how can I properly remove the GPIO isolation latch?

lurch, 2024-12-10:
Ah, well if it works with RP2350A then I guess it _should_ also work with RP2350B, so I'll leave this to Peter or Graham to investigate :slightly_smiling_face:

Octopus1633, 2024-12-23:
Hi, I would like to ask how the test is going?

peterharperuk, 2025-01-03:
I gave the host example a test and it worked on pico1 but I can't get it to work on pico2. Not immediately obvious why but I assume it's something to do with gpio not being setup properly.

arlaneenalra, 2025-01-05:
hmm .. I suspect I'm seeing something very similar but on the input side of things. I have a PIO program that waits for a to go low, then high, and it never seems to react on the rp2350. It works fine on the 2040. I'm running an RP2350A0A2 if I'm reading this right. GPIO reads show the pin changing, but PIO doesn't react at all .. odd..

peterharperuk, 2025-01-05:
I can't help you without seeing your code

arlaneenalra, 2025-01-05:
It could be unrelated..
PIO: https://github.com/arlaneenalra/CGA_LCD_Capture/blob/c63f808fc69c7abaa6c7c12ffd35135aca6897d3/src/lcd.pio
PIO Init code: https://github.com/arlaneenalra/CGA_LCD_Capture/blob/c63f808fc69c7abaa6c7c12ffd35135aca6897d3/src/main.c#L52

The signal line in question is a pulse at roughly 76hz.  The code overclocks right now, but I've tried it at the stock clock frequency and with various GPIO pin configurations. It works great on an rp2040 but doesn't do anything on the rp2350. I'm probably missing something simple... but it sounds very similar to what's described in this bug.

Octopus1633, 2025-01-06:
> I gave the host example a test and it worked on pico1 but I can't get it to work on pico2. Not immediately obvious why but I assume it's something to do with gpio not being setup properly.
@peterharperuk Thank you for your reply. I used the same firmware to test, and the result was that RP2350A tested normal, but RP2350B tested abnormal. What could be the problem?

peterharperuk, 2025-01-06:
@Octopus1633 what boards are you using with your tests?

Octopus1633, 2025-01-06:
> @Octopus1633 what boards are you using with your tests?

This is not any development board on the market. It is a board designed by me. The USB part uses the most basic circuit and is connected to the GPIO12 and GPIO13 pins of RP2350A. The pin PIO function can be used to simulate USB devices and receive data as a USB host. A stable signal can be detected on the pin.

The RP2350B board is also designed by me. The USB part also uses the most basic circuit and is connected to the GPIO7 and GPIO8 pins of RP2350B. The same program as RP2350A, only the pins are modified, but USB communication cannot be achieved. There is no stable signal on the pin.

peterharperuk, 2025-01-06:
@Octopus1633 Do you have any pico2 devices? If you do, can you see if you can get the host and device examples to work. I can only get the device examples to work on a pico2. I'm slightly suspicious that the code is suffering from an issue with the pull downs.

However, the device example works on pico2 and a pimoroni pico 2 plus which uses an rp2350b. I tested it with pins 12 and 13 so I can't reproduce your problem.

Octopus1633, 2025-01-07:
@peterharperuk I think there is a problem with the latest example of https://github.com/sekigon-gonnoc/Pico-PIO-USB. I used the 0.53 version of the example to test it on pico2, and the result shows that it is feasible as a USB device or host.

However, when compiling the 0.53 version of the example, you need to comment out the pio_sm_set_jmp_pin function definition in the sdk.

lurch, 2025-01-07:
It's probably not relevant, but just FYI: #1849 (things were in a state of flux at one point, but hopefully everything has stabilised now).

Octopus1633, 2025-01-08:
@lurch Thank you for your advice. I re-pulled the latest SDK and initialized the submodule. The test results are still the same as above.