raspberrypi/pico-sdk #2140: Trouble this Timer High Level APIs

fabianoriccardi, 2024-12-18:
I'm new about Pico SDK, so I'm just experimenting the APIs... I don't understand why this piece of code doesn't print `triggered`, unless I decomment `sleep_ms(1)`. It seems the interrupt never triggers.

Reading the [docs](https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#detailed-description-21), the alarm_callback runs in IRQ context, so I don't need to insert sleep_ms(1). Moreover, in the [example](https://github.com/raspberrypi/pico-examples/blob/master/timer/hello_timer/hello_timer.c), in the alarm_callback there is a `printf`... Is it safe calling a printf in an IRQ routine?

If I'm missing something, let me know please.

I'm using RP2350 and SDK v2.1.0.

``` C
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/timer.h"

const absolute_time_t period = 1000000;

volatile bool irqTriggered;

absolute_time_t tim = 1000000;

int64_t alarm_callback(alarm_id_t id, void *user_data) {
    irqTriggered = true;
    tim += period;
    add_alarm_at(tim, alarm_callback, NULL, false);
    return 0;
}

int main()
{
    stdio_init_all();

    irqTriggered = false;

    add_alarm_at(tim, alarm_callback, NULL, false);

    while (true) {
        //sleep_ms(1);

        if(irqTriggered){
            irqTriggered = false;
            printf("triggered\n");
        }
    }
}

```

peterharperuk, 2024-12-19:
Sorry, I can't seem to reproduce the problem. I see the printf with or without the sleep. Are you using stdio usb or stdio uart (what pico_enable_stdio_* lines do you have in your cmake file)?

> Is it safe calling a printf in an IRQ routine?
 It's safe, although it's somewhat unwise.

Adding an alarm at a specific time with add_alarm_at is a bit dangerous as it might be in the past? I would suggest adding a hard_assert that it's returning >0. And/or use add_alarm_in_ms instead.

lurch, 2024-12-19:
Also, you might want to look at https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#group_repeating_timer rather than repeatedly setting new alarms in your alarm-callback?

peterharperuk, 2024-12-19:
Also worth trying the develop branch to pick up this fix https://github.com/raspberrypi/pico-sdk/commit/969f5895aa6790a9c1af8ed31ccd06cc81fe36cf

fabianoriccardi, 2024-12-20:
> Are you using stdio usb or stdio uart (what pico_enable_stdio_* lines do you have in your cmake file)?

I'm using:

    pico_enable_stdio_uart 1
    pico_enable_stdio_usb 0

> https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#group_repeating_timer

I know this api, but I need to change the period between an activation and the next one. I know I can even use other peripherals, but now I'm just playing with this API.

I have a smaller piece of code that gives me problem:

```
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/timer.h"

int main()
{
    stdio_init_all();
    printf("Current time: %llu\n", get_absolute_time());
    printf("Current time: %llu\n", get_absolute_time());
    printf("Current time: %llu\n", get_absolute_time());
    printf("Current time: %llu\n", get_absolute_time());
    sleep_ms(100);
    printf("Current time: %llu\n", get_absolute_time());
}
```
It prints: 

```
Current time: 0
Current time: 0
Current time: 0
Current time: 0
Current time: 100000

```

It seems the timer doesn't count untill sleep_ms().

The CMakeList:

```
# Generated Cmake Pico project file

cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

# Initialise pico_sdk from installed location
# (note this can come from environment, CMake cache etc)

# == DO NOT EDIT THE FOLLOWING LINES for the Raspberry Pi Pico VS Code Extension to work ==
if(WIN32)
    set(USERHOME $ENV{USERPROFILE})
else()
    set(USERHOME $ENV{HOME})
endif()
set(sdkVersion 2.1.0)
set(toolchainVersion 13_3_Rel1)
set(picotoolVersion 2.1.0)
set(picoVscode ${USERHOME}/.pico-sdk/cmake/pico-vscode.cmake)
if (EXISTS ${picoVscode})
    include(${picoVscode})
endif()
# ====================================================================================
set(PICO_BOARD pico2 CACHE STRING "Board type")

# Pull in Raspberry Pi Pico SDK (must be before project)
include(pico_sdk_import.cmake)

project(timer C CXX ASM)

# Initialise the Raspberry Pi Pico SDK
pico_sdk_init()

# Add executable. Default name is the project name, version 0.1

add_executable(timer timer.c )

pico_set_program_name(timer "timer")
pico_set_program_version(timer "0.1")

# Modify the below lines to enable/disable output over UART/USB
pico_enable_stdio_uart(timer 1)
pico_enable_stdio_usb(timer 0)

# Add the standard library to the build
target_link_libraries(timer
        pico_stdlib)

# Add the standard include files to the build
target_include_directories(timer PRIVATE
  ${CMAKE_CURRENT_LIST_DIR}
)

# Add any user requested libraries
target_link_libraries(timer 
        hardware_timer
        )

pico_add_extra_outputs(timer)

```

peterharperuk, 2025-01-06:
This is what happens when I run your program on pico2...

Current time: 506
Current time: 2260
Current time: 4012
Current time: 5763
Current time: 107527

So I'm a bit stumped why it's behaving like that for you.

lurch, 2025-01-06:
> I'm using RP2350...

@fabianoriccardi Is that a [Raspberry Pi Pico 2](https://www.raspberrypi.com/products/raspberry-pi-pico-2/), or some other board?

lurch, 2025-01-06:
A quick fiddle with a spreadsheet reveals that the times between each of Peter's printouts above are:
1754
1752
1751
101764

closed 2025-05-29 by peterharperuk