raspberrypi/pico-sdk #2142: btstack: fails to link when using btstack-develop branch

ricardoquesada, 2024-12-19:
Btstack added new files in `develop` branch that needs to be added here: https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_btstack/CMakeLists.txt

In particular, one file that is missing is `hci_event_builder.c`.

But the problem is that there some kind of coupling (and there shouldn't be) between pico-sdk repo and btstack repo.

In theory pico-sdk should just need to say "I depened on btstack library", and btstack-lib should should be reponsible for compiling the needed files. If needed pico-sdk could add some extra files.

But the way it is structured today makes it somewhat difficult to test latest btstack versions.

I filed a similar bug on BTstac:
https://github.com/bluekitchen/btstack/issues/649

ricardoquesada, 2024-12-19:
Some examples why they should be decoupled:

- I want to use latest BTstack version because it fixes some bugs
- I found a regression. Not sure whether it is in btstack. I'm going to bisect BTstack. It should just compile without the need to modify pico-sdk CMakeLists.txt

In case it helps, I also file a bug about versions in BTstack:
https://github.com/bluekitchen/btstack/issues/640

peterharperuk, 2025-01-02:
> Some examples why they should be decoupled:

We can't decouple the sdk from btstack because there's no cmake support in btstack.

kilograham, 2025-03-18:
I forget whether CMake lets you add source files to a library from another CMakeLists.txt