raspberrypi/pico-sdk #2144: HTTPS lwip server not working

qubit999, 2024-12-21:
Despite extensive research, I found no comprehensive resources online addressing HTTPS web server implementation on Raspberry Pi Pico 2W using the Pico SDK. Most available examples focus on HTTP or unrelated TLS-client functionalities, which do not meet my requirements. 

I even submitted a feature request to the pico-examples repository, but it was closed with a suggestion to explore existing HTTP and TLS-client examples—neither of which align with my goals.

I reference this [forum post](https://forums.raspberrypi.com/viewtopic.php?p=2279368#p2279368) and its accompanying code as a starting point to fix the issue. 

However, after hours of troubleshooting, I suspect the issue lies within the Pico SDK libraries themselves. 

Specifically:
- The standard HTTP implementation (httpd_init()) functions as expected.
- Switching to httpd_inits() for HTTPS, combined with altcp_ functions (for TLS), results in errors originating from the Pico SDK libraries rather than my codebase. Notably, my code compiles without errors.
- 
At this stage, I am uncertain how to proceed, as debugging is hindered by the lack of meaningful error messages or documentation for these specific library functions.

Thank you in advance.

Best

peterharperuk, 2024-12-21:
The SDK has all the support needed to implement tls. In fact all the support is in lwip. This issue is likely to be closed as well. If you post what you've tried so far on the pico-examples issue I'll take a look at why it's not working. But implementing a tcp web server on a Pico isn't really that sensible. A better idea would be to use mqtt or even coap.

qubit999, 2024-12-21:
Hi, 

thank you for your response. Check the implementation posted on the raspberry forums (this is mine). 

My approach was to configure lwipopts.h and mbedtls_config.h. 

I also made sure to set the includes in my myhttpserver.cpp and adjusted CMakeLists.txt to include the necessary lwip and mbedtls libraries.

I replaced all tcp_ functions with altcp_.

qubit999, 2024-12-21:
UPDATE:

[{
	"resource": "/Users/alex/.pico-sdk/sdk/2.1.0/lib/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c",
	"owner": "makefile-tools",
	"severity": 8,
	"message": "mbedtls/certs.h: No such file or directory",
	"source": "gcc",
	"startLineNumber": 73,
	"startColumn": 10,
	"endLineNumber": 73,
	"endColumn": 10
}]
mbedtls/certs.h: No such file or directory

This is the only error I'm getting now.

qubit999, 2024-12-21:
```cpp
#include "myhttpserver.h"
#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"
#include "lwipopts.h"
#include <stdio.h>
#include <cstring>
#include <string>
#include <string.h>
#include <algorithm>
#include <iostream>
#include "lwip/apps/httpd.h"
#include "lwip/apps/fs.h"
#include "oled.h"

#include "certs/certs.h"

#include "lwip/altcp.h"
#include "lwip/altcp_tls.h"
#include "lwip/opt.h"
#include "lwip/apps/altcp_tls_mbedtls_opts.h"

static char stored_key1[64];
static char stored_value1[64];
static char stored_key2[64];
static char stored_value2[64];

Pico_OLED_1_3 MyHTTPServer::oled;

struct altcp_tls_config *MyHTTPServer::tls_config = NULL;

static u16_t ssi_handler(int iIndex, char *pcInsert, int iInsertLen, u16_t current_tag_part, u16_t *next_tag_part) {
    printf("SSI handler called with index: %d\n", iIndex);

    stored_key1[sizeof(stored_key1) - 1] = '\0';
    stored_value1[sizeof(stored_value1) - 1] = '\0';
    stored_key2[sizeof(stored_key2) - 1] = '\0';
    stored_value2[sizeof(stored_value2) - 1] = '\0';

    switch (iIndex) {
        case 0: // Key1
            strcpy(pcInsert, stored_key1);
            printf("SSI: Inserting Key1 = '%s'\n", stored_key1);
            return strlen(stored_key1);
        case 1: // Value1  
            strcpy(pcInsert, stored_value1);
            printf("SSI: Inserting Value1 = '%s'\n", stored_value1);
            return strlen(stored_value1);
        case 2: // Key2
            strcpy(pcInsert, stored_key2);
            printf("SSI: Inserting Key2 = '%s'\n", stored_key2);
            return strlen(stored_key2);
        case 3: // Value2
            strcpy(pcInsert, stored_value2);
            printf("SSI: Inserting Value2 = '%s'\n", stored_value2);
            return strlen(stored_value2);
        default:
            printf("SSI: Unknown index\n");
            return 0;
    }
};

err_t httpd_post_begin(void *connection, const char *uri, const char *http_request,
                    u16_t http_request_len, int content_len,
                    char *response_uri, u16_t response_uri_len, u8_t *post_auto_wnd)
{
    printf("POST request started for URI: %s\n", uri);
    return ERR_OK;
};

[... 409 lines cut ...]
    ${CMAKE_CURRENT_SOURCE_DIR}/../Pico_code/c/lib/OLED
    ${CMAKE_CURRENT_SOURCE_DIR}/../Pico_code/c/lib/Fonts
    ${CMAKE_CURRENT_SOURCE_DIR}/../Pico_code/c/examples
)

target_include_directories(myhttpserver PUBLIC
    ${CMAKE_CURRENT_SOURCE_DIR}
)

target_link_libraries(myhttpserver
    pico_stdlib
    pico_cyw43_arch_lwip_threadsafe_background
    pico_lwip_http
    pico_mbedtls
    hardware_spi
    Config
    Infrared
    OLED
    Icm20948
    LCD
    Fonts
    GUI
    examples
)
```

qubit999, 2024-12-21:
The error message looks like this:

```
In file included from /Users/alex/.pico-sdk/sdk/2.1.0/lib/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c:84:
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h: In function 'mbedtls_ssl_get_psk':
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1056:12: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1056 |     if (ssl->handshake->psk != NULL && ssl->handshake->psk_len > 0) {
      |            ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1056:43: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1056 |     if (ssl->handshake->psk != NULL && ssl->handshake->psk_len > 0) {
      |                                           ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1057:19: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1057 |         *psk = ssl->handshake->psk;
      |                   ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1058:23: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1058 |         *psk_len = ssl->handshake->psk_len;
      |                       ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1059:19: error: 'mbedtls_ssl_context' has no member named 'conf'
 1059 |     } else if (ssl->conf->psk != NULL && ssl->conf->psk_len > 0) {
      |                   ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1059:45: error: 'mbedtls_ssl_context' has no member named 'conf'
 1059 |     } else if (ssl->conf->psk != NULL && ssl->conf->psk_len > 0) {
      |                                             ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1060:19: error: 'mbedtls_ssl_context' has no member named 'conf'
 1060 |         *psk = ssl->conf->psk;
      |                   ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1061:23: error: 'mbedtls_ssl_context' has no member named 'conf'
 1061 |         *psk_len = ssl->conf->psk_len;
      |                       ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h: In function 'mbedtls_ssl_own_key':
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1141:12: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1141 |     if (ssl->handshake != NULL && ssl->handshake->key_cert != NULL) {
      |            ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1141:38: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1141 |     if (ssl->handshake != NULL && ssl->handshake->key_cert != NULL) {
      |                                      ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1142:23: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1142 |         key_cert = ssl->handshake->key_cert;
      |                       ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1144:23: error: 'mbedtls_ssl_context' has no member named 'conf'
 1144 |         key_cert = ssl->conf->key_cert;
      |                       ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h: In function 'mbedtls_ssl_own_cert':
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1154:12: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1154 |     if (ssl->handshake != NULL && ssl->handshake->key_cert != NULL) {
      |            ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1154:38: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1154 |     if (ssl->handshake != NULL && ssl->handshake->key_cert != NULL) {
      |                                      ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1155:23: error: 'mbedtls_ssl_context' has no member named 'handshake'
 1155 |         key_cert = ssl->handshake->key_cert;
      |                       ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1157:23: error: 'mbedtls_ssl_context' has no member named 'conf'
 1157 |         key_cert = ssl->conf->key_cert;
      |                       ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h: In function 'mbedtls_ssl_in_hdr_len':
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1191:12: error: 'mbedtls_ssl_context' has no member named 'conf'
 1191 |     if (ssl->conf->transport == MBEDTLS_SSL_TRANSPORT_DATAGRAM) {
      |            ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h: In function 'mbedtls_ssl_out_hdr_len':
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1202:25: error: 'mbedtls_ssl_context' has no member named 'out_iv'
 1202 |     return (size_t) (ssl->out_iv - ssl->out_hdr);
      |                         ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h:1202:39: error: 'mbedtls_ssl_context' has no member named 'out_hdr'
 1202 |     return (size_t) (ssl->out_iv - ssl->out_hdr);
      |                                       ^~
/Users/alex/.pico-sdk/sdk/2.1.0/lib/mbedtls/include/mbedtls/ssl_internal.h: In function 'mbedtls_ssl_hs_hdr_len':
[... cut ...]
/Users/alex/.pico-sdk/sdk/2.1.0/lib/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c: In function 'altcp_mbedtls_lower_poll':
/Users/alex/.pico-sdk/sdk/2.1.0/lib/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c:562:7: warning: ignoring return value of 'mbedtls_ssl_flush_output' declared with attribute 'warn_unused_result' [-Wunused-result]
  562 |       mbedtls_ssl_flush_output(&state->ssl_context);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[10/1156] Building CXX object myhttpserver/CMakeFiles/myhttpserver.dir/myhttpserver.cpp.o
ninja: build stopped: subcommand failed.
```

qubit999, 2024-12-22:
I think I am on the right way of figuring this out. Instead of using the pico-sdk built-in mbedtls library, I downloaded the library mbedtls from Github ([the zip file](https://github.com/Mbed-TLS/mbedtls/releases/)). Now I only get two errors from the mbedtls library concerning the use of entropy and time. But I will figure this out.

peterharperuk, 2025-01-02:
Please attach a link to a branch on github. Working with copy and pasted code is not ideal.

qubit999, 2025-01-02:
I fixed it.

qubit999, 2025-01-02:
For reference: https://github.com/qubit999/senfOS

peterharperuk, 2025-01-02:
Closing issue then

closed 2025-01-02 by peterharperuk