raspberrypi/pico-sdk #2147: hello_usb not working in SDK 2.1.0

tobiasmetz, 2024-12-24:
Hi,
I had problems with my raspi pico when trying to get it running with serial communications, so I got back to the hello_usb example and tried it with different SDK's under visual code.
Finally I figured out that it is not compiling properly and working under the sdk version 2.10 but well working under version 1.5.1 and 2.0.0 - so I guess it is a problem in the newest SDK.
To reproduce:
- New Project from Example hello_usb under VSCode with SDK 2.1.0
- compiling leads to errors: 
-stdio_usb_descriptors.c
    'TUD_CONFIG_DESC_LEN' undeclared here (not in a function)
    'TUD_CDC_DESC_LEN' undeclared here (not in a function); did you mean 'USBD_DESC_LEN'?
    implicit declaration of function 'TUD_CONFIG_DESCRIPTOR' [-Wimplicit-function-declaration]
    implicit declaration of function 'TUD_CDC_DESCRIPTOR' [-Wimplicit-function-declaration]
stdio_usb.c
    #warning stdio USB was configured along with user use of TinyUSB device mode, but CDC is not enabled [-Wcpp]

lurch, 2024-12-24:
I'm pretty sure that if there was _actually_ a problem with `hello_usb`, then there'd be a lot more complaints about it? :thinking: 

Are you remembering to delete your cmake build directory every time you switch SDK versions? @will-v-pi When you change SDK versions in VSCode, does the TinyUSB submodule version also update as appropriate?
I'll move this to the https://github.com/raspberrypi/pico-vscode repo for now...

peterharperuk, 2025-01-03:
Sorry, I can't reproduce this. There must be something funny about your install. I usually use Linux and it works fine. But I also used the Visual studio extension on Windows (https://marketplace.visualstudio.com/items?itemName=raspberry-pi.raspberry-pi-pico) to create a project based on hello_usb and it built and ran on Pico 2.

closed 2025-01-03 by peterharperuk