raspberrypi/pico-sdk #2148: pico_aon_timer time setting functions should add a delay on the RP2040

sneakywumpus, 2024-12-26:
We have code running on RP2040 and RP2350 that uses `pico_aon_timer` as a RTC and found that these don't work correctly on the RP2040 when a call to `aon_timer_start` is followed quickly by a call to `aon_timer_get_time`. Since `pico_aon_timer` uses the hardware RTC on the RP2040 there should be a delay after setting and before reading it.

This could be fixed by adding a delay in `aon_timer_set_time_calendar` after the `rtc_set_datetime` call, or amending the SDK documentation that one should add a delay after time setting `pico_aon_timer` function calls on the RP2040.

peterharperuk, 2025-01-02:
I'd hope we could come up with a better fix that adding a "random" delay.

lurch, 2025-01-02:
The example in section 4.8.5.2 in https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf does:
```c
// Enable RTC and wait for it to be running
rtc_hw->ctrl = RTC_CTRL_RTC_ENABLE_BITS;
while (!rtc_running()) {
    tight_loop_contents();
}
```
and section 4.8.4 mentions:
"All RTC register reads and writes are done from the processor clock domain `clk_sys`. All data are synchronised back
and forth between the domains. Writing to the RTC will take 2 `clk_rtc` clock periods to arrive, additional to the `clk_sys`
domain. This should be taken into account especially when the reference is slow (e.g. 1Hz)."

peterharperuk, 2025-01-02:
@lurch As far as I can see, the AON code does that.

kilograham, 2025-02-06:
merged into develop

PR #2156 (2025-02-06): Add a delay after rtc_set_datetime
If you call rtc_get_datetime immediately after rtc_set_datetime you get junk back. According to the datasheet "Writing to the RTC will take 2 clk_rtc clock periods to arrive". So add a delay after calling rtc_set_datetime in aon_timer_set_time_calendar.

Fixes #2148
commit 7dfbb3c9 (2025-02-06): Add a delay after rtc_set_datetime (#2156)

If you call rtc_get_datetime immediately after rtc_set_datetime you get
junk back. According to the datasheet "Writing to the RTC will take
2 clk_rtc clock periods to arrive". So add a delay after calling
rtc_set_datetime in aon_timer_set_time_calendar.

Fixes #2148
closed 2025-02-06 by kilograham