raspberrypi/pico-sdk #2149: makefsdata.py generates invalid resource name for img/rpi.png

jjonethal, 2024-12-27:
Building  on msys2 toolchain with python from mingw64
when generating "rp2350-rppico2\pico-examples\build\pico_w\wifi\httpd\generated\pico_fsdata.inc"
in array data_img_rpi_png 

    /* "/img\rpi.png" (13 chars) */
    0x2f,0x69,0x6d,0x67,0x5c,0x72,0x70,0x69,0x2e,0x70,0x6e,0x67,0x00,0x00,0x00,0x00,
makefsdata.py shall convert windows path delimiter to posix before exporting to rom file system data image.

resource is not found on subsequent request to lwip.

lurch, 2024-12-27:
Duplicate of #1793 ?

jjonethal, 2024-12-27:
no not a duplicate, this issue addresse content of data array, not the variable name. in the data array resource name also contains "\\" but web browser uses "/" as separator wich wil cause wrong resource path

jjonethal, 2024-12-27:
Correct array data shall be 
static const unsigned char data_img_rpi_png[] = {
/* "/img/rpi.png" (13 chars) */
0x2f,0x69,0x6d,0x67, **0x2f**  ,0x72,0x70,0x69,0x2e,0x70,0x6e,0x67,0x00,0x00,0x00,0x00,

/* "HTTP/1.0 200 OK" (17 chars) */

peterharperuk, 2025-01-20:
Change got merged. So closing as fixed.

PR #2150 (2025-01-07): convert windows path delimiter to posix (#2149)
Fixes #2149 
When generating lwip file system data convert windows directory separators by posix separators.
This is an issue when native directory during resource generation separator is "\\" as on windows. Running toolchain from mingw64 python3 would also preserve "\\" in resource name which would cause mismatch on resource lookup.
Fix should be portable with python 3.4 and up.
closed 2025-01-20 by peterharperuk