raspberrypi/pico-sdk #2152: CYW43 core not starting on Pico2 W

Memotech-Bill, 2024-12-29:
A call to **cyw43_arch_init_with_country** is failing.

My program is [PicoBB](https://github.com/Memotech-Bill/PicoBB). I am finding that for the Pico2 GUI build, which is using the ScanVideo library to produce VGA output on Core 1, and Core 0 running the BBC BASIC interpreter attempting to open a WiFi connection I am obtaining the following diagnostic messages:

````
[CYW43] core not in reset
[CYW43] core not in reset
[CYW43] core not in reset
[CYW43] core not in reset
[CYW43] core not up
[CYW43] HT not ready
````

and the WiFi is failing to connect.

Enabling all the diagnostics in the CYW43 driver, and comparing (diff -y) the failing case with a case in which Core 1 is idle gives the attached, where the failing case is on the left and the successful case is on the right.

[diff.txt](https://github.com/user-attachments/files/18270941/diff.txt)

Looking at the source file **pico-sdk/lib/cyw43-driver/src/cyw43_ll.c** I see the function [device_core_is_up()](https://github.com/georgerobotics/cyw43-driver/blob/cf924bb04c8984675ca0fc2178f082e404e048c3/src/cyw43_ll.c#L1260):

````
static void device_core_is_up(cyw43_int_t *self, int core_id) {
    uint32_t base = get_core_address(core_id);
    uint32_t reg = cyw43_read_backplane(self, base + AI_IOCTRL_OFFSET, 1);
    if ((reg & (SICF_FGC | SICF_CLOCK_EN)) != SICF_CLOCK_EN) {
        CYW43_WARN("core not up\n");
    }
    reg = cyw43_read_backplane(self, base + AI_RESETCTRL_OFFSET, 1);
    if (reg & AIRC_RESET) {
        CYW43_WARN("core not up\n");
    }
    // if we get here then the core is up
    CYW43_VDEBUG("core %d IS up\n", core_id);
}
````

This function seems to be testing for two possible failure conditions, but then does nothing about them other than issuing diagnostic messages. Surely it should be returning a true/false status so that subsequent code can handle any failures.

In the diagnostics attached, the following output can be seen:

````
cyw43_write_reg_u8 BACKPLANE_FUNCTION 0x1000b=0x1810		cyw43_write_reg_u8 BACKPLANE_FUNCTION 0x1000b=0x1810
cyw43_read_reg_u8 BACKPLANE_FUNCTION 0xb800=0x86010001	      |	cyw43_read_reg_u8 BACKPLANE_FUNCTION 0xb800=0x86010000
cyw43_write_reg_u8 BACKPLANE_FUNCTION 0x1000b=0x1800		cyw43_write_reg_u8 BACKPLANE_FUNCTION 0x1000b=0x1800
[CYW43] core not up					      <
core 1 IS up							core 1 IS up
````

So one of the tests is failing, but the code continues to execute as if nothing is wrong. The code eventually fails with the "HT not ready" error [here](https://github.com/georgerobotics/cyw43-driver/blob/cf924bb04c8984675ca0fc2178f082e404e048c3/src/cyw43_ll.c#L1689).

To produce the failing executable:

````
git clone --recurse-submodules https://github.com/Memotech-Bill/PicoBB.git
cd PicoBB/bin/pico
make BOARD=pico2_w ADDON=vgaboard_cut
````

Install the software on a Pico2 W on a VGA Demo Board with:

* VGA Monitor attached
* Keyboard on Pico USB connector
* Power applied to the VGA board USB connector
* Serial terminal (for diagnostics) to VGA board serial connections.

Load "bbcbasic+filesystem_gui_pico2_w_vgaboard_cut.uf2" into the pico2 W. To demonstrate failure, enter the following commands on the Keyboard attached to the pico2 W:

````
LOAD mysqldem
RUN
````

And to produce the functional executable:

````
cd PicoBB/console/pico
make BOARD=pico2_w
````

For this version no VGA board is required. Attach a USB console to the Pico2 W and run the same BASIC program.

peterharperuk, 2025-01-02:
I don't have a VGA Demo Board. Does your code work with a "normal" Pico W?
Does this Pico 2 W work ok with the pico-examples?
Am I right that this is 144000Khz?

```
set_sys_clock_khz (6 * vga_timing_640x480_60_default.clock_freq / 1000, true);
```

Memotech-Bill, 2025-01-02:
> I don't have a VGA Demo Board. Does your code work with a "normal" Pico W?

Yes, the code works on a Pico W, it only the Pico 2 that it fails on.

> Does this Pico 2 W work ok with the pico-examples?

The same network source code works on the Pico 2 with a build not making use of the second core. I am not aware of anything in either pico-examples or pico-playground that uses both scanvideo and wifi.

> Am I right that this is 144000Khz?
> 
> ```
> set_sys_clock_khz (6 * vga_timing_640x480_60_default.clock_freq / 1000, true);
> ```
Yes.

Memotech-Bill, 2025-01-02:
> I don't have a VGA Demo Board.

OK, build it this way:

````
git clone --recurse-submodules https://github.com/Memotech-Bill/PicoBB.git
cd PicoBB/console/pico
make BOARD=pico2_w GRAPH=Y SOUND=I2S
````

Install the resulting file **bbcbasic+filesystem_console_pico2_w.uf2** on a bare Pico2 W. Connect a serial terminal to the Pico USB port. The Pico pins will be busy outputting VGA video and I2S sound, but you can ignore all these. Just interact over the USB connection.

````
pi@raspberrypi:~/pico/PicoBB/console/pico $ picocom /dev/pico2
picocom v3.1

port is        : /dev/pico2
flowcontrol    : none
baudrate is    : 9600
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready

LittleFS image v2.1, Size = 1024KB, Origin = 0x10100000
BBC BASIC for Pico 2W Console v0.47, Build Jan  2 2025, USB Console, UART Console, VGA Display, Flash Filesystem, cyw43=background, /de
v/uart, Min Stack, Stack Check 4, RTC
(C) Copyright R. T. Russell, 2024
cyw43 initialisation succeded
>LOAD mysqldem
>RUN
[CYW43] core not up
[CYW43] core not up
[CYW43] HT not ready
Error -4 connecting to access point in module /lib/socklib at line 95
````

peterharperuk, 2025-01-02:
```
Connecting to MySQL server '193.62.194.222'... connected successfully

Connecting to database 'Rfam'... connected successfully
MySQL version = 5.6.36-log

Fetching database names... 12 database(s) enumerated: 'information_schema' 'Rfam' 'mysql' 'performance_schema' 'rfam_12_2' 'rfam_1
3_0' 'rfam_14_0' 'rfam_14_10' 'rfam_14_8' 'rfam_14_9' 'sj_test' 'sys'

Fetching table names... 14 table(s) enumerated: '_annotated_file' '_family_file' '_genome_data' '_lock' '_overlap' '_overlap_membe
rship' '_post_process' 'alignment_and_tree' 'author' 'clan' 'clan_database_link' 'clan_literature_reference' 'clan_membership' 'da
tabase_link'

Querying database 'Rfam' for table 'author'...

Fetching column titles... number of columns = 6
Fetching table data... number of rows read = 14

author_id  name                  last_name             initials    orcid                 synonyms
1          Ames T                Ames                  T                                 
2          Argasinska J          Argasinska            J           0000-0003-2678-2824   
3          Bachellerie JP        Bachellerie           JP                                
4          Barquist LE           Barquist              LE          0000-0003-4732-2667   
5          Barrick JE            Barrick               JE                                
6          Bateman A             Bateman               A           0000-0002-6982-4660   
7          Boese B               Boese                 B                                 
8          Boursnell C           Boursnell             C           0000-0002-3494-4451   
9          Breaker RR            Breaker               RR          0000-0002-2165-536X   
10         Brown C               Brown                 C                                 
11         Burge SW              Burge                 SW          0000-0002-2506-927X   Burge S
12         Cartinhour SW         Cartinhour            SW                                
13         Chen A                Chen                  A                                 
14         Collins JA            Collins               JA                                

Transaction completed successfully

```

Memotech-Bill, 2025-01-03:
> Reliably? I will have to try my other Pico2 W.

My second Pico2 W fails in exactly the same way.

Memotech-Bill, 2025-01-03:
Oops, an error on my part :(

 I gave the build instructions from memory, not copy & paste. Having now tried them they fail to build owing to an entirely unrelated error (building a PIO file related to sound). So I am not quite sure what you tested, possibly a version without VGA output.

The following build instructions do work, and produce an executable including VGA output:

````
git clone --recurse-submodules https://github.com/Memotech-Bill/PicoBB.git
cd PicoBB/console/pico
make clean
make BOARD=pico2_w GRAPH=Y SOUND=N
````

That is the version which is failing to connect to the network.

Memotech-Bill, 2025-01-03:
> > So I am not quite sure what you tested
> 
> I fixed the build errors.

So I am puzzled. It regularly fails on both my Pico2 W. Although I have seen it work once on one of them.

peterharperuk, 2025-01-03:
That last build command (with my changes reverted) works ok. I'll test some other devices

Memotech-Bill, 2025-01-03:
> That last build command (with my changes reverted) works ok. I'll test some other devices

Thanks.

I have pushed my version of the corrections to the sound build. None of the changes I have made affect the network code. To build with either SOUND=I2S or SOUND=PWM it would be necessary to have a board definition with the relevant pins defined.

peterharperuk, 2025-01-03:
Tried another couple of Pico 2 W devices and they all seem to get an IP address and run that program ok. What country code are you using? I'm attaching my uf2 just in case - can you see if it fails on your devices? 
[peterh_bbcbasic+filesystem_console_pico2_w.uf2.zip](https://github.com/user-attachments/files/18299797/peterh_bbcbasic%2Bfilesystem_console_pico2_w.uf2.zip)

Memotech-Bill, 2025-01-03:
Stranger and stranger.

I am using country code GB.

Your UF2 works on one of my Pico2 W. I have not yet tried the other one.

I am building an RP4B, running 64-bit Bookworm, fully updated.

````
pi@raspberrypi:~/pico/PicoBB/console/pico $ uname -a
Linux raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
````

Memotech-Bill, 2025-01-03:
My builds all include diagnostic information. Attached is an objdump of my executable. How does this compare with yours?

[bbcbasic_pico2_w.zip](https://github.com/user-attachments/files/18300047/bbcbasic_pico2_w.zip)

peterharperuk, 2025-01-03:
Ha. Yes. I was building it on a fairly old x86 ubuntu virtual machine. I can reproduce the problem when I build it on a Pi 5.

peterharperuk, 2025-01-24:
@Memotech-Bill Any chance you could test this change https://github.com/raspberrypi/pico-sdk/pull/2209
I just got it to connect ok using Oz on pico2_w but I'm building on a different Pi 5 so maybe I was just lucky.

arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205

Memotech-Bill, 2025-01-25:
> [@Memotech-Bill](https://github.com/Memotech-Bill) Any chance you could test this change [#2209](https://github.com/raspberrypi/pico-sdk/pull/2209) I just got it to connect ok using Oz on pico2_w but I'm building on a different Pi 5 so maybe I was just lucky.

I have only tested on one of my two Pico2 W, but it seems to work.

I compiled the latest version of **PicoBB** using **-Oz** and the master branch of **pico-sdk**. That failed as previously. I then replaced `cyw43_bus_pio_spi.c` with your revised version and recompiled. With the revised code, running `mysqldem.bbc` worked reliably for repeated tests.

Thank you for this.

peterharperuk, 2025-01-29:
I'm going to mark this as a duplicate of #2206 which is now fixed in develop
Thanks for your support on this issue, it was appreciated.