raspberrypi/pico-sdk #2162: RP2350A does not work properly with one PIO running three state machines

flying1425, 2025-01-06:
In the same PIO, I am using three state machines, each configured with FIFO and Scratch Registers; however, I found that the last state machine was not working properly. After migrating one of the state machines to another PIO, the problem is solved, or the last state machine works fine when its FIFO and Scratch Registers are not used at the same time. What should I do to make a PIO run all three state machines at the same time and make sure that each of them can use the FIFO and Scratch Registers properly?

peterharperuk, 2025-01-06:
Any chance of seeing the code?

flying1425, 2025-01-06:
This is the PIO program.
`

; FrontPorch:   10 clock 
; PulseWidth:   10 clock 
; BackPorch:    10 clock

.program vsync
.side_set 1 opt

; side pins ---> LCD_VSYNC_PIN(bit0)

pull block side 1 ; Pull the width to osr

.wrap_target

; FrontPorch
set x, 10 side 1
FrontPorch:
   wait 1 irq 0 side 1
   jmp x-- FrontPorch side 1

; SyncPulse
set x, 3 side 0
SyncPulse:
   wait 1 irq 0 side 0
   jmp x-- SyncPulse side 0

; BackPorch
set x, 10 side 1
BackPorch:
    wait 1 irq 0 side 1
    jmp x-- BackPorch side 1
mov x, osr side 1

ActiveFor:
    wait 1 irq 0 side 1
    jmp x--, ActiveFor side 1
.wrap

% c-sdk {

static inline void vsync_program_init(PIO pio, uint sm, uint offset, uint pin, float div)
{

    // creates state machine configuration object c, sets
    // to default configurations. I believe this function is auto-generated
    // and gets a name of <program name>_program_get_default_config
    pio_sm_config c = vsync_program_get_default_config(offset);
    sm_config_set_fifo_join(&c, PIO_FIFO_JOIN_TX);
    // Map the state machine's SET pin group to one pin, namely the `pin`
    // parameter to this function.
    sm_config_set_sideset_pins(&c, pin);

    // Set clock division (div by 5 for 25 MHz state machine)
    sm_config_set_clkdiv(&c, div);

    // Set this pin's GPIO function (connect PIO to the pad)
    pio_gpio_init(pio, pin);

    // Set the pin direction to output at the PIO
    pio_sm_set_consecutive_pindirs(pio, sm, pin, 1, true);

    // Load our configuration, and jump to the start of the program
    pio_sm_init(pio, sm, offset, &c);

    // Set the state machine running (commented out so can be synchronized with hsync)
    // pio_sm_set_enabled(pio, sm, true);
}

[... 149 lines cut ...]
    float sys_clk = clock_get_hz(clk_sys);

    uint hsync_sm = pio_claim_unused_sm(PIO_HANDLER, true);
    uint vsync_sm = pio_claim_unused_sm(PIO_HANDLER, true); //
    uint rgb_sm = pio_claim_unused_sm(PIO_HANDLER, true);   // pio_claim_unused_sm(pio2, true);

    float pio_freq = sys_clk / ((float)/*LCD_CLK_FREQ*/ 16000000.0);

    uint vsync_offset = pio_add_program(PIO_HANDLER, &vsync_program);
    uint hsync_offset = pio_add_program(PIO_HANDLER, &hsync_program);
    uint rgb_offset = pio_add_program(PIO_HANDLER, &rgb_program);

    hsync_program_init(PIO_HANDLER, hsync_sm, hsync_offset, LCD_HSYNC_PIN, pio_freq);
    vsync_program_init(PIO_HANDLER, vsync_sm, vsync_offset, LCD_VSYNC_PIN, pio_freq);
    rgb_program_init(PIO_HANDLER, rgb_sm, rgb_offset, LCD_DATA0_PIN, pio_freq);

    pio_sm_put_blocking(PIO_HANDLER, hsync_sm, LCD_WIDTH - 1);
    pio_sm_put_blocking(PIO_HANDLER, vsync_sm, LCD_HEIGHT - 1);
    pio_sm_put_blocking(PIO_HANDLER, rgb_sm, LCD_WIDTH - 1);

    // pio_sm_set_enabled(PIO_HANDLER1, rgb_sm, true);
    pio_enable_sm_mask_in_sync(PIO_HANDLER, ((1u << hsync_sm) | (1u << vsync_sm) | (1u << rgb_sm)));
}

`

peterharperuk, 2025-01-06:
What's the value of rgb_offset returned by pio_add_program?

peterharperuk, 2025-01-06:
Your programs add up to 33 instructions which exceeds the amount of memory that can fit into one PIO

closed 2025-01-06 by peterharperuk