raspberrypi/pico-sdk #2167: adc.h comments and Hardware API typos

romain145, 2025-01-07:
Is this some unfortunate Copy-paste or am I missing something? Line 109 makes reference to "RP02040".
https://github.com/raspberrypi/pico-sdk/blob/95ea6acad131124694cda1c162c52cd30e0aece0/src/rp2_common/hardware_adc/include/hardware/adc.h#L112

romain145, 2025-01-07:
line 112 is a copy-paste of line 109 (and so is line 133).

I think the correct data is on line 113?  Lines 112 and 133 should be removed.

lurch, 2025-01-08:
Line 109 (and line 129) is talking about the ADC in the RP2040 chip; see section 4.9 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

Line 112 (and line 133) is talking about the ADC in the RP2350A chip, and line 113 (and line 134) is talking about ADC in the RP2350B chip; see section 12.4 of https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf

romain145, 2025-01-08:
A vs B... My bad!
Thanks!

closed 2025-01-08 by romain145