raspberrypi/pico-sdk #2171: Possible bug in timer.h timer_time_reached

32BT, 2025-01-08:
I am under the impression that the second "hi" in the return statement of the code below should not be using the prefetched value if it is meant to guard against rollover during evaluation.

```
static inline bool timer_time_reached(timer_hw_t *timer, absolute_time_t t) {
    uint64_t target = to_us_since_boot(t);
    uint32_t hi_target = (uint32_t)(target >> 32u);
    uint32_t hi = timer->timerawh;
    return (hi >= hi_target && (timer->timerawl >= (uint32_t) target || hi != hi_target));
}

```

peterharperuk, 2025-01-08:
I don't follow your reasoning. @kilograham Does this make any sense to you?

lurch, 2025-01-09:
I've been doing some back-of-the-envelope calculations, and I don't think that this code **is** trying to deal with `timerawl` ticking over from `0xffffffff` to `0x00000000`, because the `hi >= hi_target` part of the test would still fail in that scenario anyway.

I think it's doing (basically) the same as:
```c
    (hi > hi_target || (hi == hi_target && timer->timerawl >= (uint32_t) target))
```
but just more optimised for the common-case of `hi == hi_target`. After all, a 32-bit microsecond counter will only roll-over once every 1.2 hours, and I guess the intention is for `timer_time_reached` to return as quickly as possible.

32BT, 2025-01-09:
You are probably right. In that case my apologies. I misinterpreted the code. 

Would it be impolite to suggest perhaps the following rewrite for better comprehension and futureproofing? 

```C
inline uint64_t timer_time_us_64(timer_hw_t *timer) {
    uint32_t hi = timer->timerawh;
    uint32_t lo = timer->timerawl;
    // Refetch if rollover occured while fetching 
    if (hi != timer->timerawh) {
        hi = timer->timerawh; 
        lo = timer->timerawl;
    }
	
    return ((uint64_t)hi<<32)+lo);
}

static inline bool timer_time_reached(timer_hw_t *timer, absolute_time_t t) {
    return timer_time_us_64(timer) >= to_us_since_boot(t);
}
```

The rationale for rewriting **timer_time_reached** is as follows: 
Contemporary practice suggests that most optimizations are better left to the compiler because of the vagaries of modern processors as well as future improvements like increased address sizes. How long before we get a 64bit RP MCU? The current code is not likely optimal in that case and rewriting all separate instances of old optimizations that are hard to comprehend to begin with, is probably wrought with errors.

The rationale for rewriting **timer_time_us_64** is as follows: 
Rollover while fetching is extremely rare and only reasonably occurs once. 
It therefore does not require a loop, plus optimization is probably moot. 
If an optimized loop is desired following would likely do the trick:

```C
    while (hi != timer->timerawh) {
        hi += 1;
        lo = timer->timerawl;
    }
```

My apologies in advance if the above looks like criticism. It isn't.  Consider it as enthousiasm instead. I have 4 Pico W here at home and am having a ball.

lurch, 2025-01-09:
As I already mentioned, I believe the intention is for `timer_time_reached` to return as quickly as possible (as it's probably used a **lot** inside loops). In the current implementation, it only does two register reads from the timer peripheral, whereas if you rewrote it to use `timer_time_us_64` then it'd do either 3 or 5 register reads, and every register read involves a round-trip over the APB bus (which is obviously something that the compiler can't "optimise out"). See section 2.1.3 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

> Consider it as enthousiasm instead. I have 4 Pico W here at home and am having a ball.

Good stuff! :tada:

peterharperuk, 2025-01-09:
Raise a PR if you think you can improve the code. But as suspected this is a non-issue.

closed 2025-01-09 by peterharperuk